adnanademovic/rosrust #1: Examples

andresv, 2016-08-16:
Hi,

Could you also add some basic examples (subscribe, publish) for getting quickly going.

adnanademovic, 2016-08-16:
Hi,
Thanks for the interest! I'm currently working on designing a TCPROS client (ROS is generally badly documented, so there is a lot of Wireshark fiddling in this). Once that is done, pub/sub and service/client will be a couple of hours of work to make.
After that, I'll make examples. For now you can get/set parameters (as stated in the README). The next time you get a notification from this issue, the stated features and examples will be there!

cwoodall, 2016-12-22:
If you come up with some new insights on ROS to help improve the documentation/etc you should publish them! I have been trying to find those details for a while and never actually got to sitting down with wireshark

adnanademovic, 2016-12-22:
Yeah, most of the delay in this process is caused by ROS's documentation being very lacking, which is made extra bad by the fact that a lot of elements of protocols are wildly different on the original C++ and Python implementations.

Getting all of that right is what's taking so long 😢.

mikepurvis, 2016-12-29:
In theory, the master API is documented here (and the immediate linked pages): http://wiki.ros.org/ROS/Master_API

If you're keeping notes, it would be great to have something published afterward which details how the docs diverge from reality. Also, I'm curious that you've had to reimplement xmlrpc— are they ways that ROS usage deviates from the standard that make one of the other implementations ([eg](https://libraries.io/cargo/xmlrpc)) unsuitable?

Finally, not sure how far along you are with message generation, but one thought on that is that (at least temporarily) you might be able to use a bridge like rust-cpp to reuse the serialization code avoid needing to build out all of that infrastructure.

adnanademovic, 2016-12-29:
I was less worried about Master API, and more worried about Slave API (its first method doesn't even have an explanation on what all of its various return values mean).

At the moment when I wrote the xmlrpc code there weren't any existing libraries (I started working on this several months before the linked lib even had its first release). ROS works pretty standard with XMLRPC.

Message serialization is done, I just have to look into more ROS details to choose the best way of implementing all of the genmsg bindings, and add things to the ROS build farms so that I have all the standard message types generated right in the distro. I'll do all of that once I'm done with service/client, which should hopefully be by next week!

adnanademovic, 2016-12-31:
@cwoodall @mikepurvis Since you've asked, I just ran into something right now. Apparently a client can "probe" your service, sending you this kind of a garbage header:
```json
{
  "md5sum": "*",
  "service": "/add_dem_numbers",
  "callerid": "/rosservice",
  "probe": "1"
}
```
This is not documented anywhere, and is extra annoying since I've coded things in a way that if I get a header that doesn't match, I just say no thanks, bye.

commit e8cd9f1b (2016-12-31): Add examples in the README.

This resolves #1, although the examples will still change with API
changes and extensions.
closed 2016-12-31 by adnanademovic