adnanademovic/rosrust #2: Seen jonbinney/ros_rust?

gavanderhoorn, 2016-09-28:
See [jonbinney/ros_rust](https://github.com/jonbinney/ros_rust).

adnanademovic, 2016-09-28:
I'm aware of both previous attempts of implementing a ROS Client Library. I'm not sure what I should be looking for there?

jonbinney/ros_rust doesn't fully implement any ROS feature, and is a bit badly designed structurally. Also, it looks like the project is dead.
You could even look at gch/rust-ros too. The developer paused the project to make an XMLRPC library, just so they could use it for their rust-ros project. Which is a legit way of doing things, but now both projects are dead.

AFAIK this is the only attempt that fulfills the minimum feature requirements.

closed 2016-09-28 by adnanademovic

gavanderhoorn, 2016-09-28:
I only meant to inform you. What you do with the information is up to you.

Personally I always find it valuable to look at different implementations / approaches for similar things. Perhaps naively I assumed that would work / be valuable for others as well.

adnanademovic, 2016-09-28:
Thank you very much for informing me, in any case. I've definitely browsed through these 2 repos, to decide if I'll fork from anyone, before deciding to ultimately make my own version from scratch.

You're just the first one to bring it up in an issue, so I thought I'd make a post to link anyone for future reference.

I'm sorry if my response sounded like an overreaction or rude, didn't mean to!!! 😄