adnanademovic/rosrust #10: Implement leftover Slave API methods

adnanademovic, 2016-12-31:
The [ROS Slave API docs](http://wiki.ros.org/ROS/Slave_API) aren't very informative about what is what in there, so we'll need to look further into certain details regarding "getBusStats" and "getBusInfo".

cwoodall, 2017-01-03:
I do know some people who know some ROS dev's/interned at [OSRF](https://www.osrfoundation.org/) for a summer. Let me see if I can get you some more up to date info. I just thought about that

adnanademovic, 2017-01-03:
That would be great! I've posted [a question](http://answers.ros.org/question/251215/what-do-getbusstats-and-getbusinfo-exactly-return/) on ROS Answers to see if anyone could give me any clues. There were questions before, but nobody answered them.

cwoodall, 2017-01-04:
No luck

adnanademovic, 2017-01-04:
Too bad. Still greatly appreciated. As mentioned in my question, even officially distributed client libraries just put a "TODO" there, so I did the same for now, and I've put it so it returns an error code that signals failure to handle the command. I'll probably end up solving this by analyzing the C++ and Python code, but honestly these 2 commands provide no actual functionality, so I'm in no hurry in doing so.

cwoodall, 2017-01-04:
According to my friend they think that that API is deprecated and has been replaced with NodeHandle

http://wiki.ros.org/roscpp/Overview/NodeHandles

adnanademovic, 2017-01-04:
I'm not sure how that relates to these two API methods. But it they're deprecating those things, then great. This node handle thing seems interesting by itself. I've made something similar for parameters, but this could be interesting for the ROS interface in general. But... this is a typical singleton and I hate that - one should be allowed to run 20 ROS nodes within one binary if they want to (even if that makes them incompatible with roslaunch).