adnanademovic/rosrust #9: Make error handling more sensible

adnanademovic, 2016-12-31:
Most of the error messages are just tacked on there, since the focus was getting functionality out there.

Now that functionality is there, the next step is being more clear what's going on when errors happen.

This includes getting rid of .unwrap() calls that are placed to kill a thread when we're done with it, since they are making noise on stderr.

cwoodall, 2016-12-31:
Have you seen the [error_chain](https://crates.io/crates/error-chain) crate.  Sends like a reasonable way of handling errors by automating a lot of the boilerplate.

adnanademovic, 2016-12-31:
OMG this is exactly what I was doing with needlessly complicated enums. This could work out perfectly if it's what I think it is. Thanks!

cwoodall, 2016-12-31:
I believe it was made to make the enums your making easier to make! Seems to have been developed by the rustup team

adnanademovic, 2016-12-31:
Yeah, this is perfect for this project, since most of the errors that can arise are unexpected critical errors that should be propagated to the library user, so having something like this really does the job. I'll use this ASAP.

PR #11 (2017-01-03): Revamp error handling and change some API elements
Closes #9
closed 2017-01-03 by adnanademovic