adnanademovic/rosrust #12: Write example cases like many other crates have

cwoodall, 2017-01-03:
Many crates have example cases in an examples folder which can be compiled and run right out of the gate to give users something to work with. The examples in the `README.md` file should be translatable.

adnanademovic, 2017-01-03:
I was thinking about that. What do you think good examples would be? A possible issue is the fact that we need something running on the ROS side for this to work. Should I fork roscore and rospy_tutorial/add_two_ints_service as part of a Client example, or should I just leave it to the user to launch them, and just explain it in the README, combined with error messages within the example?

I'm leaning towards the latter one, since it feels like it makes more sense

cwoodall, 2017-01-03:
I would only distribute the required rust code in the examples, and leave notes in the README about how to download and launch the appropriate ros service to make the examples work.

I would possibly consider a script for downloading and building the proper programs for the install and the appropriate ros_launch commands. That is just my opinion though.

adnanademovic, 2017-01-03:
Doing roslaunch stuff would be something for after actual integration into the ROS ecosystem (doing that right now)
I'll use rospy tutorial examples to communicate with - they come bundled in the full desktop installation of ROS

adnanademovic, 2017-01-05:
I got approval for my Rust message generation package to be pulled into ROS's respository - ros/rosdistro#13528. I'll have to wait a few days before requesting integration into message generation and ROS core, but in any case - I'll probably make these examples once I have message generation integrated into ROS core.

adnanademovic, 2017-05-11:
Examples can't work in the same repo - at least for now. Using the library at all requires message generation. There is an example by @OTL that works at [OTL/rosrust_tutorial](https://github.com/OTL/rosrust_tutorial)

It's a string message sending/receiving with a publisher and subscriber, and uses the master branch. I'll soon push out a release that contains all the major changes made since the last release!

commit 22cd0bdb (2017-12-25): Implement singleton interface

* Implemented rosrust singleton
* Cleaned up logging
* Updated README
* Finished examples (resolves #12)
closed 2017-12-26 by adnanademovic