adnanademovic/rosrust #14: Add full message generation at compile time

adnanademovic, 2017-02-20:
The release schedule for ROS packages is quite slow, and it would cause a dependency hell when interfacing with ROS itself. Besides that, there is no need for `rosrust` to even become part of standard distributed ROS, since you can compile almost everything without even having ROS on the machine.

The only exception so far are messages. I've made a `genrs` ROS package, and it's even on the ROS build farm right now, but in the end it basically utilized a bit of ROS's message generation to do a bit of preprocessing.

In the meantime, I've gotten enough knowledge about all the little details to reliably construct that code generation myself, so I'm planning to do so, and cut any dependency on ROS's release process.

IanTheEngineer, 2017-03-31:
@adnanademovic - I'm thrilled to find your support for Rust in the ROS ecosystem! Having had a hand in another client library for [rosnodejs](https://github.com/RethinkRobotics-opensource/rosnodejs) and [gennodejs](https://github.com/RethinkRobotics-opensource/gennodejs), a couple of (hopefully helpful) thoughts:
- Without `genrs` being a core dependency of [`message_generation`](https://github.com/ros/message_generation/blob/kinetic-devel/package.xml), users of `genrs` will need to have all message packages (standard and custom alike) checked out from source in their workspace to generate Rust, which is fairly burdensome.
- Becoming a dependency of `message_generation` typically only happens at the beginning of a year-long release. The Lunar release is coming up, and you might be able to sneak `genrs` in there still.
- Being a dependency of `message_generation`, means that a new release of `genrs` will cause a rebuild of the entire ROS buildfarm (because of messages). This is not a huge deal, but something to consider when considering frequent updates to `genrs`. We updated `gennodejs` twice in the past year. `rosrust` can obviously update as often as you prefer.
- With `rosnodejs` we ran into the same quandary of whether to support buildfarm message generation, or dynamically generate messages through `rosnodejs`. In the end, we decided to support both, because we figured we would have users from both the `nodejs` realm who didn't want to deal with the whole ROS distribution system, and long time ROS users who expected a client library to have messages pre-generated for them to use.

Please excuse me if you know these points already. Anyway, hope this is helpful, and feel free to reach out to me or @chris-smith if you have any question. I can't wait to try out your library (and learn Rust :)

adnanademovic, 2017-03-31:
Thanks for sharing that information, and offering a helping hand. It's much appreciated! I intentionally waited with waiting to make `genrs` a core dependency, because I was not sure if I'd commit to that path. As you see from this exact opened issue, I decided against.

I'm not far from finishing compile-time message generation, but went a bit on hiatus because I was very busy. I literally planned to work on it today, and having some feedback just extra motivated that plan!

My goal is to totally ditch `genrs`, for the exact reasons you've listed. It really would have been a big hinderance to rely on ROS's buildfarms to just get some files that I could have generated myself.

My implementation will just involve a 1 line build.rs file in the crate (which executes some compile-time code), and 1 line in the main file of the repo that uses it.

I'd recommend starting tinkering with this library upon its next release. I'll even add simple example apps to run against real ROS nodes.

Thanks again for the feedback!

IanTheEngineer, 2017-03-31:
It seems you've already got `genrs` ready to go, why not support both local generation and buildfarm generation? Once you've decided on a class-generation format, `genrs` shouldn't need much in the way of future updates. That would allow your ROS environment users to skip the compile step for all message packages (which for large projects can amount a few minutes to build hundreds of message packages).

adnanademovic, 2017-03-31:
Generating messages should be blazingly quick. The code for using `genrs` and just generating at compile time is identical from the library consumer side. Also, I consider disk I/O to be the bottleneck for message generation, and this new system would perform half the amount of disk input, and the same amount of output.

adnanademovic, 2017-05-13:
This has been resolved with the `rosrust 0.5` release, and can be seen in the demo referenced in the README

closed 2017-05-13 by adnanademovic