adnanademovic/rosrust #20: How is progress?

surfertas, 2017-08-18:
Hi adnanademovic.

I was wondering how development of rosrust was progressing? Is there anywhere that I can help? What are the plans from here?

adnanademovic, 2017-08-18:
Hi!

I've been on quite a long pause, doing other things. And funnily enough, yesterday I started fiddling with things again.

Currently, I'm making a standalone `xml-rpc` server, based on `serde` and the new `tokio`-based `hyper` version. I hope to be done with that over the weekend, and then plug it in here later!

If you're familiar with ROS's protocols, feel free to poke around wherever you want. Just know that I'm replacing the `rosxmlrpc` portion of the codebase with this thing I'm making, so you can ignore that area.

I'm not familiar with your Rust and ROS experience, so I wouldn't know where the best place would be to help. One thing I have 0 experience with are procedural macros. If you're familiar with those, message generation is currently implemented as a build script. If you're interested, I can help you get into that system and explain how things currently work.

surfertas, 2017-08-18:
Hi Adnan, thanks for the response. In response to my familiarity with Rust and ROS, its quite limited. I have been a user, and not a developer, but keen on helping where I can as Rust and ROS are both tools that I would like to use actively. 

If you can point me in the right direction, I can take a look at current implementation, and see if conversion to procedural macros is something within my capacity. (If not, I can always get help from Rust experts around me.)

If you think of something more trivial, I am also happy to help there as well. 

Best Regards,

Tasuku

surfertas, 2017-08-18:
I am reviewing procedural macros, and appears well above my capabilities. Again if there is something more trivial I am happy to help in any way I can. Thanks Adnan.

adnanademovic, 2017-08-18:
Yeah, they are quite a thing to deal with.

I'll work a bit with the code over the weekend, and I'll actively look for easy tasks to get acclimated with the code. If I find anything, I'll create more issues and ping you in this thread!

Thanks for the interest!

surfertas, 2017-08-18:
Thats great. Thanks a lot. Have a nice weekend.

surfertas, 2017-08-19:
Adnan, sorry am playing around with rosrust, I ran into this situation.

When passing a topic name such as "cmd_vel" to ros.publish("cmd_vel").unwrap, I had no issues in building and publishing to the cmd_vel topic.

The issue arises when I try to pass a topic with a namespace associated, such as "/raspimouse/diff_drive_controller/cmd_vel" in which I get the following error. 

ERROR:rosrust::rosxmlrpc::server: Server handler error: Issue while reading XML-RPC data, for request
Caused by:Failed to parse parameters
Caused by:Failed to parse XML RPC parameter
Caused by:Value node should contain one node representing its data type

My current solution is using a python script to remap from cmd_vel to /raspimouse/diff_drive_controller/cmd_vel, which allows me to send velocity commands to a robot running in simulation (Gazebo).

I am most likely just not passing the topic name in the right form. I looked through the api source code and saw that you were running tests for such names, but the solution was not immediate thus my inquiry,

Thanks and sorry for the troubles.

adnanademovic, 2017-08-19:
The XML RPC section of the code was slapped together very sloppily - and that's the portion I'm working on a rewrite for. Combined with the fact that this portion is also poorly documented, things like this can happen.

With the error that was thrown, your ROS node received a malformed request. It's possible that that's Gazebo's doing.

But, this is the easiest way to test the culprit: keep using a python script, but remap from `/some/path/cmd_vel` to `/raspimouse/diff_drive_controller/cmd_vel`, and have the Rust node publish to `/some/path/cmd_vel`.

If it fails, that means that there is an issue with `rosrust`. If it succeeds, it means that Gazebo's XML-RPC messages are the issue.

surfertas, 2017-08-19:
Hi Anand, thanks a lot for the response. I tested as instructed, and the mapping works, thus it appears that Gazebos XML-RPC messages are the issue.

adnanademovic, 2017-08-19:
Good to know! Even though it's annoying that those issues exist.

Just to document and explain what the cryptic error message means:

This is the [XML-RPC spec](http://xmlrpc.scripting.com/spec.html).

If you look at any of the examples, you'll notice the general rule that things are wrapped in `<value>...</value>`, and that this node is never empty. In fact, inside that node is always exactly one node, showing the data type (`<string>...</string>`, `<array>...</array>`, and other). If you look at the error now, you'll notice that it says this is missing.

Suffice to say, this should never happen.

russkel, 2017-10-08:
Hi all,
Quick question, seeing as ROS 2.0 is around the corner, what's the plan in regards to compatibility with that version?
Russ

adnanademovic, 2017-10-08:
@russkel 

If you look at [ROS 2 client library instructions](https://github.com/ros2/ros2/wiki/ROS-2-Client-Libraries), you can see the [comparison to ROS 1](https://github.com/ros2/ros2/wiki/ROS-2-Client-Libraries#comparison-to-ros-1).

The bindings will most probably be a whole different repo, like `rosrust2` or something. The reason is that it really is a change from the ground up - literally. ROS 2 will be implemented by every programming language making calls to a common C library - [RCL](http://docs.ros2.org/beta1/api/rcl/). This will make the ROS client very easy and quick to implement, once the RCL is finalized.

I have no intentions of pursuing a day 1 release of the Rust version by supporting all the betas, since I'm doing all of this for free and have a very limited amount of time allotted to doing this.

russkel, 2017-10-10:
Thanks for your detailed reply @adnanademovic :).

adnanademovic, 2017-12-26:
@surfertas I don't know if you care anymore, but I fixed the incompatibility with some message sources, including (hopefully) Gazebo.

Sorry, was a busy year!

adnanademovic, 2017-12-26:
@surfertas I don't know if you care anymore, but I fixed the incompatibility with some message sources, including (hopefully) Gazebo.

Sorry, was a busy year!

P.S. I released 0.6, take a look, it's full of major changes.

surfertas, 2017-12-29:
Got it, I will take a look. Thanks a lot, and hope you are having a nice end to the year.