adnanademovic/rosrust #26: auto-deserialization of parameters broken

astraw, 2017-10-02:
It seems the auto-deserialization of parameters is broken as documented in the README:

```
error[E0277]: the trait bound `msg::std_msgs::UInt64: rustc_serialize::serialize::Decodable` is not satisfied
  --> /home/astraw/src/mycrate/src/ros.rs:96:24
   |
96 |             param.get::<msg::std_msgs::UInt64>().unwrap().data
   |                   ^^^ the trait `rustc_serialize::serialize::Decodable` is not implemented for `msg::std_msgs::UInt64`

error: aborting due to previous error

error: Could not compile `mycrate`.

```

(This is on current master. Using `param.get_raw()` instead of `param.get::<msg::std_msgs::UInt64>()` works fine.)

Without digging in, my guess is that this broke in the transition to serde.

Relatedly, it would be good to get some tests, even simple ones, on the parameter api, as I didn't find any at the moment.

adnanademovic, 2017-10-02:
That is correct. The issue should be fixed once `xmlrpc-api-rework` is merged, as that one uses serde as well, as shown [here](https://github.com/adnanademovic/rosrust/blob/xmlrpc-api-rework/src/api/master.rs#L99-L101).

Though the branch is feature complete in an ideal world, a lot of ROS things are poorly documented, so I'll have to handle all the weird cases that some language implementations tend to create (like instead of RPC calls doing `method(param1, param2, param3)`, they do `method([param1, param2, param3])`.

In any case, I hope to merge it very soon.

And I agree about tests, that's why I made it the #7 issue.

adnanademovic, 2017-12-26:
@astraw 

I've released a total rehaul. The changes you've added are still in 0.5.1 (I know it's anti-semver, 0.X releases tend to be more like 0.major.minor anyway).

But I think you'll like the new interface much more.

Also, a lot of bug fixes, and more are on the way for sure.

adnanademovic, 2017-12-26:
Regarding the parameters - parameters have nothing to do with message types. What you can do is:
`param.get::<u64>()` rather than using `std_msgs::UInt64`. I mean, you can do even that now, but that would just be another generic structure. You can look at the example I added to see usage.

closed 2017-12-26 by adnanademovic

astraw, 2017-12-26:
Great that you are picking this up again.

I will check things out and file any issues I come across with the refactored code. Things are currently working reasonably well for us with rosrust, so I have a good baseline for real-world testing, I think.