adnanademovic/rosrust #27: no messages received from c++publisher

beschaef, 2017-12-06:
Hi Adnan,

Our Goal is to publish a message in C++ and subscribe it in Rust. So our Steps we actually take are:
- We are publishing in an c++ Node some Car infos. The message includes the header, position, speed and acceleration of the car. 
- Then we try to subscribe this Topic with our Rust Node. 
with "rostopic echo" we can see that there are messages published, but not our rust node.
- On the other hand if we subscribe to the topic with an c++ oder python node we get the data. We also get the messages if we publish to this topic with a rust node. 

If we add the env_logger we get the following messages:

```
ERROR:rosrust::api::slavehandler: Failed to connect to publisher 'http://Lenovo:34875/': Issue while reading XML-RPC data, for response
Caused by:Failed to parse parameters
Caused by:Failed to parse XML RPC parameter
Caused by:Failed to parse array's children
Caused by:Expected a node named 'value' with 1 child
ERROR:rosrust::api::ros: Failed to subscribe to all publishers of topic '/cars/a/car_connector/PositionUpdate': Issue while reading XML-RPC data, for response
ERROR:rosrust::api::slavehandler: Failed to connect to publisher 'http://Lenovo:34875/': Issue while reading XML-RPC data, for response
Caused by:Failed to parse parameters
Caused by:Failed to parse XML RPC parameter
Caused by:Failed to parse array's children
Caused by:Expected a node named 'value' with 1 child
ERROR:hyper::server: request error = TooLarge
```

If we change the branch in the Cargo.toml to
`rosrust = { git = "https://github.com/adnanademovic/rosrust.git", branch = "xmlrpc-api-rework" }`

We get the followed messages if we try to subscribe to the topic

```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Error(XmlFormat(Msg("Failed to parse XML-RPC response.")), State { next_error: Some(Expected token &XmlEvent::Characters(ref name) |
&XmlEvent::StartElement { name: OwnedName { local_name: ref name, .. }, .. }, found EndElement(value)), backtrace: None })', /checkout/src/libcore/result.rs:906:4
note: Run with `RUST_BACKTRACE=1` for a backtrace.
benjamin@Lenovo:~/catkin_ws/src/anki/racing$ ./target/debug/racing_ki 
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Error(XmlFormat(Msg("Failed to parse XML-RPC response.")), State { next_error: Some(Expected token &XmlEvent::Characters(ref name) |
&XmlEvent::StartElement { name: OwnedName { local_name: ref name, .. }, .. }, found EndElement(value)), backtrace: None })', /checkout/src/libcore/result.rs:906:4
note: Run with `RUST_BACKTRACE=1` for a backtrace.
```

adnanademovic, 2017-12-06:
Can you give me the msg file you're using, or the name of the message type if it's a standard one?!
So I can try and recreate it

beschaef, 2017-12-07:
This is the Message i use.

 std_msgs/Header header
uint8           location_id
uint8           road_piece_id
float32         offset_from_road_center_mm
uint16          speed_mm_per_sec
uint8           parsing_flags

uint8           last_recv_lane_change_cmd_id
uint8           last_exec_lane_change_cmd_id
uint16          last_desired_horizontal_speed_mm_per_sec
uint16          last_desired_speed_mm_per_sec

bool            reversed_parsed
bool            reversed_driving

adnanademovic, 2017-12-10:
I diagnosed the issue. As usual, it was due to lack of standard compliance. And the C++ client turns out to be tremendously bad.

Just for reference, this is the XMLRPC spec: http://xmlrpc.scripting.com/spec.html

This is what the C++ client sent:
``` xml
<?xml version="1.0"?>
<methodResponse>
  <params>
    <param>
      <value><array><data>
        <value><i4>1</i4></value>
        <value></value> <!-- illegal, cannot be left empty -->
        <value><array><data>
          <value>TCPROS</value> <!-- illegal, string tags are mandatory, similar to the i4 tags below -->
          <value>myhostname</value> <!-- illegal, same issue -->
          <value><i4>43975</i4></value>
        </data></array></value>
      </data></array></value>
    </param>
  </params>
</methodResponse>
```

I'll try to fix this today, and hope that the C++ client doesn't do even weirder stuff

adnanademovic, 2017-12-10:
@beschaef If you want to continue with work until this is fixed, you can do this, it will 100% work:

* assume you're publishing the message type `foo_msg/Bar` to topic `buz`
* publish the C++ publisher to `buz_ugly`
* launch: `rostopic echo /buz_ugly | rostopic pub /buz foo_msg/Bar`

The command creates a bridge between the C++ and Rust interface. But, I hope I'll have this fixed by tomorrow, so you don't ever need to fiddle with this!

adnanademovic, 2017-12-10:
The `xmlrpc-api-rework` branch should have that fixed now. I tested it with the message, and it worked. Let me know if it worked out for you!

beschaef, 2017-12-11:
Thank your very much, now it's working

closed 2017-12-11 by beschaef