adnanademovic/rosrust #31: Integrate with the catkin build system

adnanademovic, 2017-12-26:
It would be greatly useful to build upon `catkin_make` invocations, and to be callable from `rosrun <package> <node>`. The latter would by extension allow running from launchfiles.

The current solution is creating shell scripts that run `cargo run ...` for the binary in question, and putting them in the `/scripts` subfolder. That's similar to how python scripts are already being run.

This isn't very urgent, since this hacky solution works for now.

neachdainn, 2019-07-03:
Related: #113 

I don't think we'll be able to do anything better than hacky unless Catkin learns about Rust.

neachdainn, 2019-07-07:
I did have an idea that might be viable: we could create a `cargo-catkin` that implements all of the changes specified in #113. It could set up the `.cargo/config` file and then, for each crate in the workspace, set up a simple `CMakeLists.txt` and `package.xml` base on the metadata available in the crate's `Cargo.toml`. It wouldn't be a perfect solution but it might be the best possible way to integrate without having to make modifications to Catkin itself.

I'm fairly busy at the moment, but this is definitely on my list of things I would like to do.

OTL, 2019-07-08:
I think it is great idea!

pmirabel, 2020-07-21:
> I'm fairly busy at the moment, but this is definitely on my list of things I would like to do.

Hi, I'm currently working on a ROS project involving some rust code... Would be happy to help you develop this feature.

neachdainn, 2020-07-21:
While this is still on my list of projects I would like to eventually address, I have since moved on to a project that currently does not involve ROS. If I do find time to work on this, I'll probably try to introduce a level of Rust support in CMake (maybe based on the [Devolutions attempt][1]) as that solves this issue without being limited to ROS.

If you want to work on this yourself, I'd be more than happy to help with any questions you have.

[1]: https://github.com/Devolutions/CMakeRust

pmirabel, 2020-08-17:
I cu

pmirabel, 2020-08-17:
I will let you know when I get back to this subject, I have to successfully compile a rust node first...  and stuck with some novice comile errors.

kjeremy, 2020-09-30:
This might also be a potential starting point: https://github.com/AndrewGaspar/corrosion

neachdainn, 2020-10-02:
> This might also be a potential starting point: https://github.com/AndrewGaspar/corrosion

I have tried that out for a small project and it's absolutely delightful so far. If #112 is solved, I think that would put this crate in a *very* good place.

kjeremy, 2020-10-07:
It looks like corrosion has a minimum cmake version of 3.12 while melodic has a much lower minimum cmake.

neachdainn, 2020-10-07:
Upgrading CMake is [easy][1] and is otherwise moot without this crate declaring a MSRV.

[1]: https://apt.kitware.com/

kjeremy, 2020-10-07:
I accidentally posted this in #113 instead of here:

After adding the cmake ppa and updating my cmake I was able to build/install corrosion. Then I could do:

```cmake
cmake_minimum_required(VERSION 3.0.2)
project(publisher_node)

find_package(catkin REQUIRED)
find_package(Corrosion REQUIRED)

corrosion_import_crate(MANIFEST_PATH Cargo.toml)

catkin_package()

corrosion_install(TARGETS ${PROJECT_NAME}
  RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)
```

This builds but did not install my `publisher_node`. I can't figure out how to get `rosrun` to see my package until I build with `catkin_make install`.

icefoxen, 2021-11-23:
@kjeremy Thank you for that cheat sheet, it works very well for me using ROS Noetic and the `catkin` command (not `catkin_make`).  Even works with a workspace containing multiple binaries, I just change the `corrosion_install()` line to:

```cmake
corrosion_install(TARGETS binary1 binary2
  RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)
```

ModProg, 2023-04-30:
My CMakeLists looks like this and seems to work without a workspace:
```cmake
cmake_minimum_required(VERSION 3.0)
project(<package>)

find_package(catkin REQUIRED)
catkin_package()

add_custom_target(<package>
   ALL
   COMMAND cargo build --release --manifest-path ${CMAKE_CURRENT_SOURCE_DIR}/Cargo.toml --target-dir ${CMAKE_BINARY_DIR}/cargo
   COMMAND ${CMAKE_COMMAND} -E copy ${CMAKE_BINARY_DIR}/cargo/release/<package> ${CMAKE_RUNTIME_OUTPUT_DIRECTORY}/<package>
   COMMENT "Building <package>!!!"
)
```