adnanademovic/rosrust #32: Special semantics for Header need to be implemented

adnanademovic, 2017-12-26:
The header have a timestamp and sequential ID set if they are not specified.

Thus, if the timestamp equals zero, read the current time and set it to that.

If the sequential ID equals to zero, read it from an atomic counter and then increment it.

More [here](http://wiki.ros.org/msg#Header).

OTL, 2017-12-26:
> Thus, if the timestamp equals zero, read the current time and set it to that.

I think auto filling of header.stamp is not good idea, personally.
Is there any implementation in which do this? I think header.stamp=0 is used to explain
special case.

> If the sequential ID equals zero, read it from an atomic counter and then increment it.

This should be implemented, but header.seq is deprecated.

https://discourse.ros.org/t/is-the-sequence-number-in-header-message-deprecated/1528

adnanademovic, 2017-12-26:
Thanks for the info! I'm working of implementing this right now, since I realized it's simple and quick to do. It should be ready as a 0.6.1 in less than an hour hopefully.

I considered defaulting the header to the timestamp, but that would punish people that want to initialize message objects using literals. So I decided to have `publisher.send(message)` fix the header in the consumed message and send it.

The seq is 5 minutes of work, so I might as well implement it.

PR #33 (2017-12-26): Set header stamp and seq when zeros
Fixes #32.
commit 32a9bf61 (2017-12-26): Set header stamp and seq when zeros

Fixes #32.
closed 2017-12-26 by adnanademovic