adnanademovic/rosrust #34: no messages received in c++ subscriber

OTL, 2017-12-31:
Now it is possible to publish from C++, (#27) but c++ listener does not work.

I'm talking about below case.

* Publisher: examples/publisher.rs
* Subscriber: roscpp_tutorials/listener

```
roscore
```

```
rosrun roscpp_tutorials listener
```

```
cd rosrust
./target/debug/publisher
```

`rostopic echo /chatter` works fine, because it is python(?).

adnanademovic, 2017-12-31:
I'll just document what my debug process was.

Once I tapped into roscpp's debug log, it said this:

```
Node: /listener
Time: 14:45:14.590936769 (2017-12-31)
Severity: Debug
Published Topics: /rosout

XML-RPC call [requestTopic] didn't return an array

Location:
/tmp/binarydeb/ros-kinetic-roscpp-1.12.12/src/libros/xmlrpc_manager.cpp:XMLRPCManager::validateXmlrpcResponse:215

----------------------------------------------------------------------------------------------------
```

Using Wireshark I realized that [for the XML-RPC Slave API](http://wiki.ros.org/ROS/Slave_API), even though it's documented as returning 3 parameters, `roscpp` actually wants 1 array parameter with 3 fields. I'll do that fix throughout the Slave API, since I'm sure that it wants that kind of response for every method.

adnanademovic, 2017-12-31:
I tested roscpp_tutorial/rospy_tutorial listener/talker/add_two_ints_service/add_two_ints_client, and it worked. Fix should be in 0.6.2 in a few minutes

closed 2017-12-31 by adnanademovic

OTL, 2018-01-01:
great quick fix!!

commit 57ec2e9f (2017-12-31): Fix connection to roscpp

Fixes #34

The reasons have been listed in the mentioned issue