adnanademovic/rosrust #36: pub sub not works in `/use_sim_time` `true` and `/clock` is published

OTL, 2018-01-01:
pub sub example does not work when

* `/use_sim_time` param is true
* `/clock` is published correctly

## How to reproduce this problem

```
roscore
rosparam set /use_sim_time true
rostopic pub -s -r 1000 /clock rosgraph_msgs/Clock 'now'
./target/debug/publisher
./target/debug/subscriber
```

## `roscpp_tutorials` works fine in this situation.

```
roscore
rosparam set /use_sim_time true
rostopic pub -s -r 1000 /clock rosgraph_msgs/Clock 'now'
rosrun roscpp_tutorials talker
rosrun roscpp_tutorials listener
```

OTL, 2018-01-01:
FYI: What actually I want to do is,

publish `control_msgs/FollowJointTrajectoryActionGoal.msg` from `rosrust` to gazebo simulator.

I used 

```
rostopic pub -s -r 1000 /clock rosgraph_msgs/Clock 'now'
```

instead of

```
roslaunch ur_gazebo ur5.launch
```

adnanademovic, 2018-01-01:
This should also be fixed now

closed 2018-01-01 by adnanademovic

adnanademovic, 2018-01-01:
Deployed in 0.6.3

commit f94f0a6a (2018-01-01): Fix clock waiting

Fixes #36.

The simulated clock subscriber would lock the clock with an infinite
loop if there is nothing looking at the time.

Another issue was that running a node before initializing the clock
would not wait until non-zero reads for doing rates and sleeps.