adnanademovic/rosrust #37: Unregister topic does not work

OTL, 2018-01-01:
Unregister of topics does not seem to be working correctly.

You can check this when you kill publisher by `Ctrl-c`.

```
roscore
./target/debug/publisher #[Ctrl-c]
```

```
$ rostopic list
/chatter
/rosout
/rosout_agg
```

This should be 

```
$ rostopic list
/rosout
/rosout_agg
```

adnanademovic, 2018-01-01:
Interesting. I guess I should make it so that ros::init creates an intercept to the SIGINT signal, and let rosrust::spin() break and rosrust::ok() return false upon catching it.

adnanademovic, 2018-01-01:
The development branch should have a fix for this now.

The caveat is that now if you want to have that infinite loop, it has to be done via `while rosrust::is_ok() {...}` or `rosrust::spin()`, but I think that is a reasonable expectation to have in the first place.

closed 2018-01-01 by adnanademovic

adnanademovic, 2018-01-01:
Deployed in 0.6.3

commit ed1e87d9 (2018-01-01): Fix typo for unregistering subscriber

Together with the previous commit, fixes #37