adnanademovic/rosrust #41: problems with service in rust

astraw, 2018-02-12:
My testing of rosrust 0.6.4 has most things working, but I'm hitting a couple issues with services. First is an error (`"Broken pipe (os error 32)"`) printed in service calls. This happens when the service is called by the client example in `examples/serviceclient`.

Here is how I run the service:

```
rosrust/examples/serviceclient$ cargo run --bin service
    Finished dev [unoptimized + debuginfo] target(s) in 0.0 secs
     Running `/home/astraw/src/rosrust/target/debug/service`
[INFO @ examples/serviceclient/src/service.rs:23]: 4 + 5 = 9
ERROR:<unknown>: Broken pipe (os error 32)
[INFO @ examples/serviceclient/src/service.rs:23]: 4 + 5 = 9
ERROR:<unknown>: Broken pipe (os error 32)
```

Here is how I run the client:

```
rosrust/examples/serviceclient$ cargo run --bin client -- 4 5
    Finished dev [unoptimized + debuginfo] target(s) in 0.0 secs
     Running `/home/astraw/src/rosrust/target/debug/client 4 5`
[INFO @ examples/serviceclient/src/client.rs:35]: 4 + 5 = 9
[INFO @ examples/serviceclient/src/client.rs:48]: 4 + 5 = 9
```

Although the rust client example evidently successfully calls the service, I think my second issue may be related: the standard ROS tools cannot interact with the service without error:

```
rosservice info add_two_ints
Node: /add_two_ints_server
URI: rosrpc://localhost:37153
ERROR: Unable to communicate with service [/add_two_ints], address [rosrpc://localhost:37153]
```

Or 

```
rosservice call add_two_ints 4 5 
ERROR: Unable to communicate with service [/add_two_ints], address [rosrpc://localhost:37153]
```

adnanademovic, 2018-02-12:
I'll try to Wireshark it as soon as I can. I could have sworn I tried out all combinations by hand, but I guess it reached a point where automated tests will be necessary.

I'll post some updated info as soon as I have an idea of what the issue is.

astraw, 2018-02-12:
I now think these are two different issues.

I discovered that disabling IPv6 on the machine in question allows the `rosservice` calls to succeed. (This can be done temporarily on linux with `echo "1" > /proc/sys/net/ipv6/conf/all/disable_ipv6`.) The [rospy service demo](http://wiki.ros.org/ROS/Tutorials/WritingServiceClient%28python%29) works with or without IPv6 enabled, so I think rosrust can be fixed to make this work. I have been setting the env var `ROS_HOSTNAME=localhost`, which resolves to both `::1` and `127.0.0.1`.

I haven't gotten anywhere with the "broken pipe" issue.

astraw, 2018-02-12:
I created a new issue for IPv6, let's keep this issue about the broken pipe error.

astraw, 2018-02-12:
I renamed this issue and updated my original description to keep this issue focused on the broken pipe issue.

I did notice that things seem to work. Even so, my logs are filling up with the broken pipe issue, which makes it harder to deal with other issues in my code. :)

adnanademovic, 2018-02-12:
At the moment I'm not at a computer with ROS on it to apply fixes, but you can comment this line out to silence that: [service.rs#L143](https://github.com/adnanademovic/rosrust/blob/3250a0efa289d030113ddee48932cd5e2165e38f/rosrust/src/tcpros/service.rs#L143) (if I'm not mistaken).

adnanademovic, 2018-02-13:
Turns out a broken pipe is just a scary way of saying that the other party closed the connection, [based on the docs](https://doc.rust-lang.org/std/io/enum.ErrorKind.html#variant.BrokenPipe).

I'll create a similar patch to your PR #47, without the loop modification.

closed 2018-02-13 by adnanademovic

astraw, 2018-02-13:
Great, thanks. Your soluting is working for me in my initial testing.

commit 806356e0 (2018-02-13): Silence broken pipe errors on connection close

A broken pipe or connection reset is expected upon a closed
connection, based on the docs:
https://doc.rust-lang.org/std/io/enum.ErrorKind.html#variant.BrokenPipe

This fixes #41 and closes #47