adnanademovic/rosrust #46: service fails on machine with IPv6 enabled

astraw, 2018-02-12:
As explained in #41, which I originally thought was likely from a single bug, the rosrust service demo (started with `rosrust/examples/serviceclient$ cargo run --bin service`) can fail with standard ROS tools:

```rosservice info add_two_ints
Node: /add_two_ints_server
URI: rosrpc://localhost:37153
ERROR: Unable to communicate with service [/add_two_ints], address [rosrpc://localhost:37153]
```

Or

```
rosservice call add_two_ints 4 5 
ERROR: Unable to communicate with service [/add_two_ints], address [rosrpc://localhost:37153]
```

I discovered that this does not happen when IPv6 is disabled on the machine in question. The tests were done with the env var `ROS_HOSTNAME=localhost`.

adnanademovic, 2018-11-10:
I'm not sure how to replicate this issue.

clynamen, 2018-11-23:
I think setting the ipv6 loopback address in /etc/hosts is enough to reproduce the problem.

No error messages are shown, and no ros message are published. I am not sure wheter ROS is compatible with ipv6, in this case it is an bug in rosrust. 
Otherwise, maybe it would be good to raise an error when the hostname is resolved?

lparadasanmartin, 2020-11-12:
Hi Adnan, Andrew,
    Working on the integration of rosrust on my ROS based robot I faced this very same error.
    I don't know if this can help you on finding the root cause of the problem Adnan but when not being able to communicate with the service I noticed that I'm also unable to fetch the service info :
`rosservice info /my_node/hello_world 
Node: /my_node
URI: rosrpc://localhost:36081
ERROR: Unable to communicate with service [/my_node/hello_world], address [rosrpc://localhost:36081]`

    The error indeed disappears as soon as I disable IPV6 following Andrew's recommendation: `echo "1" > /proc/sys/net/ipv6/conf/all/disable_ipv6`

    Hope this helps, it would be cool to have this working while IPV6 is enabled too :)

Luis