adnanademovic/rosrust #48: feature request: publisher latching

astraw, 2018-02-14:
There is one feature we find useful in ROS not implemented in rosrust: publisher latching. A latched publisher keeps a copy of the last sent message and transmits it to any newly connecting subscribers. It is described a little more in [the ROS 0.7 release announcement](http://www.ros.org/news/2009/07/ros-07-released.html).

I looked at the publisher/subscriber code in rosrust, but this does not look entirely simple to add, at least not without being more familiar with the code base.

adnanademovic, 2018-02-15:
Shouldn't be too hard to do. I'll consider implementing it during this weekend.

adnanademovic, 2018-04-30:
I was thinking of adding the limited message queues before making another release, that would contain this feature.

I've made another example within `examples/pubsub` just for toggling this feature. And the latching is done by default.

Sorry for the long delay, it was a busy time for me!

closed 2018-04-30 by adnanademovic

astraw, 2018-05-01:
Many thanks, this is great. I have one concern I think is important, which is that setting latching true by default is contrary to the normal expectation in ROS. Typically when you get a message on a topic, you know that this has just been sent and is thus low-latency. I'm concerned that latching by default will break this expectation. I think it is best if rosrust adheres to ROS conventions and expectations as much as possible.

adnanademovic, 2018-05-01:
I agree. I flipped it.

astraw, 2018-05-01:
Wonderful. We will test this soon. Thanks again.

PR #49 (2018-04-30): Publisher latching
Resolves #48