adnanademovic/rosrust #53: very slow pub/sub

durka, 2018-07-10:
I'm sorry for the vagueness of this issue. Basically, I wrote something that I thought should be very simple (receive messages on a topic and publish them all to another topic) and it turned out really slow, much slower even than rospy.

Meat of the code in each language (gist follows):

```rust
let img_pub = RefCell::new(rosrust::publish("/out")?);
img_pub.borrow_mut().set_queue_size(Some(10));

let idx = AtomicUsize::new(0);
let _img_sub = rosrust::subscribe("/in",
                                  move |v: msg::sensor_msgs::Image| {
                                      let i = idx.load(Ordering::SeqCst) + 1;
                                      ros_info!("{}", i);
                                      idx.store(i, Ordering::SeqCst);

                                      img_pub.borrow_mut().send(v).unwrap();
                                  })?;

rosrust::spin();
```

```py
class sub_cb:
    def __init__(self, pub):
        self.pub = pub
        self.i = 1

    def __call__(self, data):
        print self.i
        self.i += 1
        self.pub.publish(data)

def main():
    pub = rospy.Publisher('/out', Image, queue_size=10)
    sub = rospy.Subscriber('/in', Image, sub_cb(pub))
    
    rospy.spin()
```

[Here is the full code.](https://gist.github.com/durka/70175783a971cbd4cc26005e4ed6173d)

I'm testing this using a bag that plays messages at a rate of 15 Hz. The Python node can keep up with this without a problem. However, the Rust node forwards messages with a delay of 1-2 minutes (!) and only manages 3 Hz.

I'm wondering if you have some ideas on why the Rust bindings are slower here, or what I'm doing differently in Rust vs Python.

adnanademovic, 2018-11-10:
I'd need the bag itself to see what the issue is more precisely.

adnanademovic, 2018-11-10:
I've identified the issue and created an example to test it. I'll look into what the performance issue could be.

adnanademovic, 2018-11-10:
You should try running in release mode (`cargo run --release` if you're using cargo). That resolved it for me. Procedures for reading streams were taking up massive chunks of time and this resolved that.

Please let me know if this fixed it!

PR #93 (2019-04-09): Resolve various outstanding issues
Fixes #92, fixes #71, fixes #69, fixes #66, fixes #53, fixes #45.
closed 2019-04-09 by adnanademovic