adnanademovic/rosrust #55: services and publishers not usable from remote machine

astraw, 2018-08-13:
I have been testing rosrust in a multi-machine environment but I cannot get services or publishers to work from a remote machine. Here is my testing procedure:

#### machine 1 (IP addresss 10.4.66.28), terminal window 1
    export ROS_MASTER_URI="http://10.4.66.28:11311"
    roscore

#### machine 1 (IP addresss 10.4.66.28), terminal window 2
    export ROS_MASTER_URI="http://10.4.66.28:11311"
    cd examples/serviceclient
    cargo run --bin service

As of current master (ffbf1baf6cd652a4ed738a8b68d05cc32a136381), this works from machine 1:

 #### machine 1 (IP addresss 10.4.66.28), terminal window 3
    export ROS_MASTER_URI="http://10.4.66.28:11311"
    rosservice call /add_two_ints 2 3
    # prints `sum: 5`

However, this does not work:

 #### machine 2 (IP addresss 10.4.66.24)
    export ROS_MASTER_URI="http://10.4.66.28:11311"
    rosservice call /add_two_ints 2 3
    # prints `ERROR: Unable to communicate with service [/add_two_ints], address [rosrpc://machine1:41745]`

The situation is similar for the publisher. In that case, I have been testing with `cargo run --bin publisher` in `examples/pubsub` and `rostopic echo /chatter`.

adnanademovic, 2018-11-10:
Just as a litmus test: have you tried doing this with `rostopic`?

Place `rostopic pub -r 5 /foo std_msgs/String "<some message>"` on both machines, and also doing `rostopic echo /foo` from both to see responses. Of course, use different messages to see which machine is publishing.

When using ROS_MASTER_URI via IP address rather than hostname it just prints messages from the same machine, but no communication between the machines.

If you want to just use IP addresses, you should set each machine's `ROS_HOSTNAME` to its IP address, so the master to `export ROS_HOSTNAME=10.4.66.28`, and the master to whatever its IP address is.

adnanademovic, 2018-11-10:
I tried on my laptop and desktop, by setting `ROS_HOSTNAME` to their IP addresses, and `ROS_MASTER_URI` the same way you did.

I tried all the combinations:

I ran `rostopic echo (..)` and `cargo run --bin subscriber` on both machines, and then tried both `rostopic pub (..)` and `cargo run --bin publisher` on either of the machines - all 16 combos working.

I ran `rosservice client (..)` and `cargo run --bin client (..)` on both machines, with `cargo run --bin service` already running on one of the machines - all 8 combos working.

I tried all of this on the `proc-macros` branch, which I plan to merge and publish as `0.7` as soon as I clear out as many issues as feasible.

chachi, 2019-01-08:
I've been able to replicate this, as well. @adnanademovic does it work for you to publish from a rosrust node and echo on another machine?

I've traced the problem to the various places `TcpListener::bind` is called. The `address` variable which gets passed down to these calls is the `self.hostname` member of the `Ros` struct. Binding this way prevents outside connections.

The simplest fix is to instead use `0.0.0.0` as the bind address. This allows any other machine to connect. 

It seems rospy and roscpp do it slightly differently, with the undocumented feature where if `ROS_HOSTNAME` or `ROS_IP` (or their `__*` counterparts) are set to `localhost` or `127.*` then they will bind to localhost, otherwise they bind to `0.0.0.0`. I'll prep a PR to behave the same.

adnanademovic, 2019-01-08:
@chachi 
As I've mentioned before, I tried every combination. As mentioned before, all 24 possible combinations of using `rostopic` and `rosrust`. Have you tried to see if `rostopic` works in cases where `rosrust` doesn't? Please look at my first response to the issue to see what I mean.

chachi, 2019-01-08:
Gotcha. I suspect setting your `ROS_HOSTNAME` directly to the machine's IP is providing the different behavior then. If you set it to your machine's hostname instead, then that will resolve to 127.0.0.1 and, I believe, not allow external connections.

chachi, 2019-01-08:
To elaborate a bit more, by setting `ROS_HOSTNAME` to the machine's IP means the `bind` call will listen for any traffic destined for that IP, which works. By setting `ROS_HOSTNAME` to the machine's hostname or leaving it blank and allowing the `resolve::system_hostname()` call to get the hostname, the dhcp resolves that name to a local IP and `bind` only allows traffic originating from the local machine.

Hope that's helpful. I'm testing the changes to bring rosrust in line with rospy/roscpp on my end now, they're pretty minor.

chachi, 2019-03-03:
This should be fixed and closeable now. cc @astraw

astraw, 2019-03-03:
That's great. I will try to check this out soon, but it'll likely be a couple weeks. Thanks to you both.

I'm closing this issue now. Please reopen if there is a problem.

closed 2019-03-03 by astraw