adnanademovic/rosrust #65: Let me help you to add more messages

DaFox416, 2019-01-10:
Hi, I'm working in a project that needs sensor_msgs/NavSatFix message, and currently is not implemented, so I want to help you to add more message types. I don't know how, so I need you to teach me.

Thanks.

garethellis0, 2019-02-03:
Just popping in here because I was browsing issues to possibly work on.

You might want to start by looking in `rosrust_codegen/src/msg_examples/sensor_msgs/msg/`. There are already examples there for `Image.msg` and `Imu.msg`. I think those should provide sufficient examples for all the types you would need to implement `NavSatFix`?

Hope this helps!

DaFox416, 2019-02-05:
Thank you, man! I'm going to check that examples and see if I can add the NavSatFix.

levilovearch, 2022-05-04:
Hi, I'm wondering whether it will be possible to support customised messages?