adnanademovic/rosrust #85: Info about export ROSRUST_MSG_PATH missing

MattsSe, 2019-03-04:
I don't have ros installed on my machine (macos), so after cloning and running `cargo build` I encountered:
```
error: proc macro panicked
 --> rosrust/src/msg.rs:1:1
  |
1 | rosmsg_include!(rosgraph_msgs / Clock, rosgraph_msgs / Log, INTERNAL);
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  |
  = help: message: called `Result::unwrap()` on an `Err` value: Error(Msg("Could not find requested message in provided folders: rosgraph_msgs/Log"), State { next_error: None, backtrace: None })
```
So I had a look in the README especially the Message Generation section, but I couldn't figure it out at first. After looking in the travis.yml I found the relevant info to `export ROSRUST_MSG_PATH=rosrust_codegen/src/msg_examples`, at least I think it is, because afterwards it compiles.

Is this the right way to do it or did I miss something?

Perhaps this info should be in the README directly, maybe in a build instruction section like:

```
git clone git@github.com:adnanademovic/rosrust.git
cd rosrust
export ROSRUST_MSG_PATH=rosrust_codegen/src/msg_examples
cargo build

```

Since I don't know if this is a non issue, I didn't submit a PR for that, but if you're used to set the env variable and this info is beneficial I'm happy to do it. 

**rust version: 1.34.0-nightly**

adnanademovic, 2019-03-04:
In general, the idea is to have the targeted ROS version installed on the used machine. That way one can use that version's messages to generate the code. We can't make any assumptions about the actual format in which the messages are provided. The environment variable you mentioned is mostly for testing purposes in an isolated environment - testing on new hardware without needing to install all of ROS, for CI (as you mentioned, in `travis.yml`).

At the end of the day, the example messages are there to have reliable tests. If you want to interact with ROS properly, you'd need ROS installed. But I guess it can be argued, since `rosrust` works without ROS installed by faking the messages, that this option should be made more clear. But even in that case, `CMAKE_PREFIX_PATH` is the actual, canonical way of providing paths to ROS message definitions.

MattsSe, 2019-03-04:
that makes sense.
Thanks for elaborating!
Closing this then.

closed 2019-03-04 by MattsSe