adnanademovic/rosrust #92: published messages not seen by some tools

6xzo, 2019-03-21:
A very simple publisher, such as the chatter example in the readme, publishes messages that don't seem to be seen by all tools. `rostopic echo /chatter` sees them, but `rostopic hz /chatter` and `rosbag record /chatter` do not. I noticed this while investigating that a LabView node also does not observe the messages. Publishing the same with `rostopic pub` or a Python publisher and all tools observe the messages. Tried with Ubuntu 16 Mate/kinetic on a raspberry Pi, Ubuntu 18 Mate/melodic on x64, and even with RosOnWindows/melodic, same result everywhere.

adnanademovic, 2019-04-02:
Turns out it's tools that upon requesting a topic request a wildcard message type, instead of a fixed message type.

adnanademovic, 2019-04-09:
The `wip` branch should have that fixed now.

closed 2019-04-09 by adnanademovic

6xzo, 2019-04-09:
Thank you. I will test it on my systems.

PR #93 (2019-04-09): Resolve various outstanding issues
Fixes #92, fixes #71, fixes #69, fixes #66, fixes #53, fixes #45.