adnanademovic/rosrust #100: Services shouldn't need Sync

dunmatt, 2019-05-15:
I'm trying to write a service, but it's complaining that my (manually written...) ServicePair struct isn't Sync.  I can certainly go add an Arc<Mutex<, but I really shouldn't have to, the function I send in to `rosrust::service` should be given ownership of the new message, so `Send` should be sufficient.

adnanademovic, 2019-05-15:
Can you give me an example of how the code looks?
I'm not fully clear on what the issue is.

dunmatt, 2019-05-15:
let (io_state_tx, io_states) = channel();
    rosrust::service("ur_driver/set_io", move |io: ur_msgs::SetIoRequest| {
        Ok(ur_msgs::SetIoResponse { success: io_state_tx.send(io).is_ok() })
    }).unwrap();

adnanademovic, 2019-05-15:
Thanks.
I'll look into this tomorrow. You are right - I don't see any reason for that specific code not to compile.

chachi, 2019-05-16:
I believe the where clause on the callback below is partly responsible for the Sync requirement, though it’s possible that’s necessary for the downstream threading model. 

impl Service {
    pub(crate) fn new<T, F>(
        master: Arc<Master>,
        slave: Arc<Slave>,
        hostname: &str,
        bind_address: &str,
        name: &str,
        handler: F,
    ) -> Result<Self>
    where
        T: ServicePair,
        F: Fn(T::Request) -> ServiceResult<T::Response> + Send + Sync + 'static,

adnanademovic, 2019-05-16:
Turns out the limitation is not without a reason!
A service is able to handle more than one request at once. Thus, multiple threads should be able to call this callback at the same time.
Turns out one `mpsc::Sender` cannot be accessed from multiple threads at once, due to explicitly implementing `!Sync`.
I'd recommend using [crossbeam channels](https://docs.rs/crossbeam/0.7.1/crossbeam/channel/) for handling functionality that mpsc was handling. It has the same interface, with `Sync` implemented on senders, and with multiple receivers supported.
[This](https://github.com/adnanademovic/rosrust/blob/master/rosrust/src/tcpros/service.rs#L77-L100) is the piece of code that requires `Sync` - anything downstream does not have the `Sync` requirement anymore.
A new reference to the handler is handed to another thread for each new node that requires the service.

dunmatt, 2019-05-16:
Ah, that makes perfect sense!  Don't know why I didn't think about concurrent requests...

Thanks for you help!

closed 2019-05-16 by dunmatt