adnanademovic/rosrust #101: Status / purpose of rosrust_codegen crate is unclear

tstellanova, 2019-05-18:
This repo is listed as the repository for both `rosrust` and `rosrust_codegen` crates and the README appears identical for both crates.  It does not appear to be possible to use `rosrust_codegen` separately from `rosrust`.  (If it is possible, I sincerely would appreciate some minimal docs on how to use it standalone.)

I don't know the reasons why the `rosrust_codegen` crate was published, but short of unpublishing it, perhaps the README for the `rosrust_codegen` crate could be updated to indicate that it's deprecated and `rosrust` should be used directly instead?

adnanademovic, 2019-05-19:
`rosrust_codegen` is a direct dependency of `rosrust`. It's only responsibility is implementing the `rosmsg_include!` macro, since it's a procedural macro, and procedural macros need to be implemented in a crate of their own.

Though I will consider updating the README, the new README will probably just be "procedural macros for <rosrust link>".

Thanks for the concern!

tstellanova, 2019-06-02:
Since `rosrust_codegen` is a procedural macro crate, consider:
-  Moving the source code for `rosrust_codegen` to its own repo instead of including within the `rosrust` repo
- Make it possible to use the `rosrust_codegen` without `rosrust`.  Currently attempting to do this (by forcing a dependency on just the `rosrust_codegen` directory) results in:

```
error[E0433]: failed to resolve: use of undeclared type or module `rosrust`
 --> src/lib.rs:8:5
  |
8 |     rosmsg_include!(sensor_msgs/Image,sensor_msgs/Imu,geometry_msgs/TransformStamped);
  |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ use of undeclared type or module `rosrust`
```

The functionality provided by `rosrust_codegen` to generate rust types from ROS messages seems very useful on its own.

closed 2019-06-02 by tstellanova

adnanademovic, 2019-06-03:
Again, `rosrust_codegen` is a key part of `rosrust` and the only reason it's a separate crate is technical limitations of the crates system.
Depending on `rosrust_codegen` alone does not make sense since the generated messages implement traits from `rosrust`.
It's all similar to how `serde` and `serde_derive` interact.