adnanademovic/rosrust #105: Document how to add custom message folders

neachdainn, 2019-06-06:
It seems that there is a mechanism for pointing `rosrust_codegen` to message definitions that does not depend on having sourced the Catkin setup file. Documentation on how to do this would be very helpful and help in making this crate a viable method for running ROS nodes on systems that aren't Ubuntu.

neachdainn, 2019-06-06:
I *think* I have figured out how to do this via looking through the code but I haven't fully tested it. If I manage to get it working I will try to submit a merge request when I have time.

neachdainn, 2019-06-06:
I did not realize that #85  had been opened (and closed) previously. Apologies for the duplicate issue, but I would like to comment on the following: 

> In general, the idea is to have the targeted ROS version installed on the used machine. That way one can use that version's messages to generate the code.

One of the major appeals of this crate is that I do not have to have ROS installed on the machine. It allows me to develop and test of something compiles without having to drop into an Ubuntu VM. Often, the only messages that are being used are ones that are specific to the project and/or the standard/common messages. While they technically do depend on the specific ROS versions, the standard messages [haven't changed in eight years][1] and the common messages seem to be just as stable. I think the benefits far outweigh the downsides here.

> But even in that case, CMAKE_PREFIX_PATH is the actual, canonical way of providing paths to ROS message definitions.

While this issue isn't tied to a specific variable name, I believe `ROSRUST_MSG_PATH` to be a more Rusty way of handling this situation. The `CMAKE_PREFIX_PATH` can be confusing to navigate and it seems odd to reference an unrelated build system, particularly if people are not using CMake to build their code or are otherwise unfamiliar with it (I also have yet to find a good, standard way of building Rust libraries with CMake). In addition, it requires a structure to the folder that does not make a lot of sense outside of a CMake build. It may also help to point out that other Rust crates, such as `env_logger` do utilize a Rust specific environment variable.

[1]: https://github.com/ros/std_msgs/tree/groovy-devel/msg

adnanademovic, 2019-06-07:
While I don't 100% agree that `ROSRUST_MSG_PATH` is necessary, I see no harm in keeping it in existence for now.
Anybody who doesn't need it doesn't need to care about its existence, and people who do need it have a method equivalent to the `CMAKE_PREFIX_PATH` approach, without the need to reference anything related to `cmake`.

PR #106 (2019-06-07): Document ROSRUST_MSG_PATH
Closes #105 

It actually worked well once I had figured out what the code was doing. Definitely did not take as long as I was worried it might.
closed 2019-06-07 by adnanademovic