adnanademovic/rosrust #111: "Failed to remove excess data from message queue"

neachdainn, 2019-06-21:
Is this actually an error? I am getting it fairly consistently (two or three times per second) and I am willing to spend a little bit of time trying to fix it.

adnanademovic, 2019-06-22:
First of all, for future reference, this is where it fires:

https://github.com/adnanademovic/rosrust/blob/71d01ad24eced5d7d94318ee210e498c9518983e/rosrust/src/util/lossy_channel.rs#L46-L54

The method `try_recv` fails if there are no more senders and the channel is empty. We always have a sender at this line of code (a sibling field to this receiver), and the channel can be only empty if a proper receiver actually emptied it between the loop condition and call.

For the code you are running - I assume the queue size on your publisher/subscriber is tiny. I'd recommend putting a larger number to prevent this issue in the future!

With that in mind - I'd remove this error message and add counters for counting dropped messages, and instances where a drop was attempted and failed. This is still an issue where a warning that this is happening could be useful.

I'll put more consideration into this.

AntoniosBarotsis, 2023-06-28:
Could this maybe be added to the docs of the method itself? I scoured through various methods that take `queue_size` as an argument but didn't see it mentioned somewhere (unless I somehow missed it). 

I could also add it myself, just let me know where you think it makes the most sense to do so :)

neachdainn, 2023-10-24:
Just going through my notification box on Github and noticed this was still open, so I'll just pop in to say that all of my projects in the last four-ish years filter out log messages from `rosrust` because of this, but that's such a trivial thing to do that it doesn't actually impact me. That being said, this is also a trivial fix if there's a clear log level that fits this kind of message (unless there's a more preferable path going forward.

> For the code you are running - I assume the queue size on your publisher/subscriber is tiny. I'd recommend putting a larger number to prevent this issue in the future!

For the record, I don't want to do this because my subscriber actually only needs the messages at a smaller rate than the publisher is providing them, so adding additional iterations/logic just to drop messages didn't seem ideal.

> Could this maybe be added to the docs of the method itself? I scoured through various methods that take queue_size as an argument but didn't see it mentioned somewhere (unless I somehow missed it).

It probably should be added to the docs but this is a core aspect of how ROS1 PUB/SUB works and I suspect that there was an assumption of some level of familiarity. In order to really cover this in the docs, you'd probably need to add information about it to `Subscriber`, `Ros::subscribe`, and `singleton::subscribe` (plus all the variants).