adnanademovic/rosrust #113: How to use roslaunch with rosrust?

OTL, 2019-07-02:
I want to use rosrust binary with roslaunch.

How do you integrate rust program with roslaunch?
It is possible to copy the binary to somewhere like `catkin_ws/develop/lib/PACKAGE/`, but is there good integration with cargo?

neachdainn, 2019-07-03:
My Catkin integration method is to use a CMake file like the following:

```cmake
cmake_minimum_required(VERSION 3.0)
project(my-project)

find_package(catkin REQUIRED)
catkin_package()

add_custom_target(my-project
	ALL
	COMMAND cargo build --release -p my-project
	COMMAND ${CMAKE_COMMAND} -E copy ${CMAKE_BINARY_DIR}/cargo/release/my-project-binary ${CMAKE_RUNTIME_OUTPUT_DIRECTORY}/my-project-binary
	COMMENT "Building my Rust library"
)
```

I also have the following in `.cargo/config`:

```toml
[build]
target-dir = "build/cargo"
```

I also have all of the Rust libraries use a shared workspace to prevent building duplicate libraries.

neachdainn, 2019-07-03:
I think anything beyond that will probably require modifying Catkin to know about Rust and that will probably be hindered by Cargo not providing a way to specify output directories.

OTL, 2019-07-03:
Thank you! It seems great!
I did not think of using catkin.

Currently I'm using below workspaces style.

```
~/rust/: for all rust programs contains rosrust programs
~/catkin/: for ros without rust
```

Your idea is below right?
```
~/rust/: only for pure rust excludes rosrust
~/catkin/: for ros includes rosrust
```

Are you comfortable about it?

neachdainn, 2019-07-03:
I have all my Rust code for this project inside of Catkin "packages" even if it doesn't use ROS just to make it easier for my collaborators to use the libraries without having to deal with anything Rust. That is, I don't have to explain to them that there is a `cargo build --release` step - they just run `catkin_make` or `catkin build` and it just works.

OTL, 2019-07-03:
I see. I'll use your idea!

Thanks a lot!

OTL, 2019-07-04:
Hi @neachdainn 
i'm trying the integration.

I created metapackage which include everything, and placed it ~/catkin_ws/src/the_meta_package/
And I needed ~/catkin_ws/src/Cargo.toml.

It is like

```
[workspace]
members = [ "src/the_meta_package/each_package1", .... ]
```
And 
each_package1/CMakeLists.txt is like you showed in this thread.

Is this what you mean?
Could you give me some advice if you don't mind?

neachdainn, 2019-07-04:
My workspace roughly looks like this:

```
catkin_ws
|--- Cargo.toml
|--- .cargo
|    |--- config
|--- src
     |--- rust_pkg_1
     |     |--- Cargo.toml
     |     |--- CMakeLists.txt
     |     |--- package.xml
     |     |--- src
     |     |     |--- lib.rs
     |--- rust_pkg_2
     |     |--- Cargo.toml
     |     |--- CMakeLists.txt
     |     |--- package.xml
     |     |--- src
     |     |     |--- lib.rs
     |--- non_rust_pkg
           |--- usual_cpp_stuff
```

Where the root Cargo.toml contains the workspace information.

OTL, 2019-07-05:
Great! thank you!

I'm using `catkin build`, then 

```
	COMMAND ${CMAKE_COMMAND} -E copy ${CMAKE_BINARY_DIR}/cargo/release/my-project-binary ${CMAKE_RUNTIME_OUTPUT_DIRECTORY}/my-project-binary
```

is not working. ${CMAKE_BINARY_DIR} seems to be `catkin_ws/src/my-project/build`.

I'm using `${CMAKE_SOURCE_DIR}/../` something instead of ${CMAKE_BINARY_DIR}.

Do you know about it?

neachdainn, 2019-07-06:
I don't know why that wouldn't work. My only guess is you might need to make sure that you declare the `catkin_package` before you add the custom target?

OTL, 2019-07-08:
It have `catkin_package`. But it is not related to rosrust, i'll dig into catkin by myself.

OTL, 2019-07-09:
It seems that the difference between `catkin_make` and `catkin build`.
`catkin build` does not have root project.

I guess that you are using `catkin_make`. When you create some tools in the future, i'm happy if you consider `catkin build` users.

neachdainn, 2019-07-09:
Is `${CMAKE_RUNTIME_OUTPUT_DIRECTORY}` still the correct folder with `catkin build`? I think I know how to make this work as long as that variable is correctly set. If it isn't, I'll have to get more creative.

neachdainn, 2019-07-09:
I have found that I have similar issues with `catkin_make_isolated`.

neachdainn, 2019-07-09:
I have dug through the CMake variables and it seems there is no way to detect if you are running `catkin_make`, `catkin_make_isolated`, or `catkin build`.

nobuyuki-tomizawa, 2020-03-11:
( Thank you guys for nice information. It's super nice if we could generate package.xml & CMakefile.txt :-) with cargo sub-command or something. )

elsuizo, 2020-08-17:
Hi all!!
i got a program for automate the solution of @neachdainn 

https://github.com/elsuizo/ros-project-generator

I am using a template library maybe it is not necessary. 

`package.xml` file generation is missing

pmirabel, 2020-09-18:
Hello! 
At the price of an environment variable, I was able to keep my traditional git repo hierarchy (pushing only package, not whole `catkin_ws` needed because of the root `cargo.toml`). I use `catkin_make_isolated` and I added a roslaunch example to make it "plug and play" :) 
You can find it here : https://gitlab.com/pmirabel/rosrust-boilerplate 
Please tell me if you see some improvements, I am still not comfortable with all of this !

kjeremy, 2020-10-07:
After adding the cmake ppa and updating my cmake I was able to build/install corrosion. Then I could do:

```cmake
cmake_minimum_required(VERSION 3.0.2)
project(publisher_node)

find_package(catkin REQUIRED)
find_package(Corrosion REQUIRED)

corrosion_import_crate(MANIFEST_PATH Cargo.toml)

catkin_package()

corrosion_install(TARGETS ${PROJECT_NAME}
  RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)
```

This builds but did not install my `publisher_node`. I can't figure out how to get `rosrun` to see my package until I build with `catkin_make install`.

neachdainn, 2020-11-25:
I have not tested this out, but something like this might work:

```cmake
add_custom_command(TARGET ${PROJECT_NAME} 
    POST_BUILD
    COMMAND ${CMAKE_COMMAND} -E copy "${${PROJECT_NAME}_binary_dir}/${PROJECT_NAME}" "${CMAKE_RUNTIME_OUTPUT_DIRECTORY}/${PROJECT_NAME}"
)
```
Basically, after the build is done copy the binary file from the build directory into the appropriate devel directory. `rosrun` doesn't do anything clever when it is run, so copying the final executable to the correct directory should just work. I do something similar to this in my code and it works, though I don't know if it will break anything e.g. installing.