adnanademovic/rosrust #114: Message macro doesn't work for all package names

neachdainn, 2019-07-03:
Trying to generate a message that exists in a package is not possible as the macro directly uses the package name as an identifier. This usually isn't too bad since the typical ROS package name has the "package_name" but causes problems if the package uses a "package-name" style.

A possible solution for this would be to have it approximate the behavior of Cargo and substitute an `_` for all `-` in the name.

adnanademovic, 2019-07-17:
Do `roscpp` or `rospy` have issues with the generated code when packages are named that way?

ROS has specific naming guidelines for many things, including packages the last time I checked.

neachdainn, 2019-07-18:
I knew about the guidelines but I had just assumed they were guidelines and not enforced. Turns out, they are. Perhaps the better solution then is to improve the error message by pointing the user to [REP 144]?

[REP 144]: http://www.ros.org/reps/rep-0144.html

commit 076357a0 (2019-11-07): Validate package names in codegen

Ensures package names follow ROS REP 144.

Fixes #114.
closed 2019-11-07 by adnanademovic