adnanademovic/rosrust #121: Getting data out of subscribers

djrakita, 2019-11-07:
Hi!  I am pretty new to Rust, so I apologize if this ends up being a naive question.  But, I haven't seen this addressed in issues or examples thus far.

I totally understand how subscribers are intended to work here -- set up a callback as a subscriber and process the incoming data in that callback.  Pretty straight forward and pretty much the exact same as rospy and roscpp.  However, it's unclear to me how to actually get the data OUT of the callback in order to process it elsewhere.  All of the examples provided only do simple processing (e.g., printing incoming values), but never show what it would look like to do this data storage step.  

Given that the expected callback types are Fn's and not FnMut's, it's not possible to mutate the environment outside of the callback and, for example, store values in some struct outside of the callback.  Would it be possible to change the callback type to FnMut instead of Fn to make this possible?  Or, maybe I'm missing something easy.  Any help would be appreciated, thanks!

adnanademovic, 2019-11-07:
Callbacks are running on different threads, so you need to do thread synchronization yourself. How you do that is in no way related to `rosrust` itself.

But I'll give you two easy approaches to collecting the callback back into the main thread:

The first one is channels - https://doc.rust-lang.org/std/sync/mpsc/index.html. The first example shows you how to put a sender in a callback, and then you can call the receiver until the sender disappears - which happens when the subscriber goes out of scope

The second one is just an `Arc<Mutex<YourStructure>>`, where main has one instance, the callback has the other, and each thread locks it to read/write to it. Here are some examples: https://doc.rust-lang.org/std/sync/struct.Mutex.html

adnanademovic, 2019-11-07:
I'll close this because it isn't actually an issue with the library.

closed 2019-11-07 by adnanademovic

djrakita, 2019-11-07:
Ah gotcha, thanks for the quick response!