adnanademovic/rosrust #123: How does support for custom messages work.

tdejager, 2019-11-09:
Hi,

This is more of a question than an issue. I wanted to test the rosrust library with our own message types. The workspace is set-up, and everything is sourced. It seems that the `include_message` macro looks at the following locations:

```
        /home/tdejager/ros/melodic/smart_robotics/devel/share
          /opt/ros/melodic/share
          /home/tdejager/ros/melodic/smart_robotics/devel/../src
          /opt/ros/melodic/../src
```

So that is the `share` folder for the catkin workspace and the source folder. However, I think that the messages for packages in the user workspace are not copied to the `share` folder. At least, I can't seem to find them there. Also as for the source folder, the required package is not located directly in the root, but in a subdirectory.

Am I missing something? Could you maybe share some info on how the include macro is supposed to work?

adnanademovic, 2019-11-09:
Where are your messages located? The actual `.msg` files I mean. Are they in `/home/tdejager/ros/melodic/smart_robotics/src/my_package/msg/SomeMessage.msg`?

tdejager, 2019-11-09:
They are actually in a subfolder for our (mono)-repo, so `/home/tdejager/melodic/smart_robotics/sr-common/sr_msgs/msg/Foo.msg`

adnanademovic, 2019-11-09:
Oh I never considered if `catkin_make` can find packages that aren't directly located in the src folder. I guess the codegen should search through the subfolders as well. I'll try to get to that soon.

andresv, 2019-11-09:
I have following structure:
```
|-common/msg/SomeMsg.msg
|-rustnode
```
All custom messages are under `common/msg`. I use `ROSRUST_MSG_PATH` to specify where those messages are.
So following line is in `rustnode/CMakeLists.txt`
```
COMMAND ROSRUST_MSG_PATH=${CMAKE_CURRENT_SOURCE_DIR}/.. cargo build --release --target-dir ${CMAKE_BINARY_DIR}/cargo/
```

In `rustnode` in can include them as:
```
use rosrust;

rosrust::rosmsg_include!(
                common/SomeMsg,
);
```

tdejager, 2019-11-10:
Wow, thank you both for the answers. And thank you for making this project btw @adnanademovic, you have done a lot of great work :)

@andresv Thanks for that solution, I'll try it out. Is that also how you are building the rust project in the catkin workspace?

@adnanademovic If you give me some pointers where to start, I've checked the proc-macro code a bit. I could try to make a PR for that?

andresv, 2019-11-10:
@tdejager, yes, this how my rust nodes are integrated with catkin. So basically others almost even do not know that there are rust nodes. They just run `catkin_make` as always.

OTL, 2019-11-11:
+1 for searching subdirectories as @adnanademovic says.
I need a lot of ROSRUST_MSG_PATH to handle my/others meta packages.

commit 477d516f (2019-11-25): Traverse subfolders when looking for dependencies

Fixes #123
closed 2019-11-25 by adnanademovic