adnanademovic/rosrust #134: Failed to build example, trait Copy not implemented

romainreignier, 2020-03-25:
Hi!
I wanted to test rosrust which seems like a really cool project but I did not manage to build a simple node from the examples.

I am creating a new project with `cargo new --bin subscriber_node` and then replace the `main.rs` content with the [`examples/examples/subscriber_node.rs`](https://github.com/adnanademovic/rosrust/blob/0ded7f0b3a2e98474ccc8fadc8416a08456409b2/examples/examples/subscriber_node.rs) content and in my `Cargo.toml` I have the following:

```toml
[dependencies]
rosrust = "0.9"
rosrust_msg = "0.1"
env_logger = "0.7.1"
```

My ROS Melodic workspace is sourced in my terminal and when I issue `cargo build`, I get the following error:

```
error[E0277]: the trait bound `std_msgs::String: std::marker::Copy` is not satisfied
 --> /home/rre/projects/rust_ws/subscriber_node/target/debug/build/rosrust_msg-ee3a560c2745cf1d/out/messages.rs:3:9
[...]
  |         |
  |         the trait `std::marker::Copy` is not implemented for `std_msgs::String`
  |         in this macro invocation
  |
  = note: the `Copy` trait is required because the repeated element will be copied

error: aborting due to previous error

For more information about this error, try `rustc --explain E0277`.
error: could not compile `rosrust_msg`.

To learn more, run the command again with --verbose.
```

Did I missed something in my setup?

I have tried with the publisher node example but the result is the same.

This is my current Rust installation:
```
rustup show
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/rre/.rustup

installed toolchains
--------------------

stable-x86_64-unknown-linux-gnu
nightly-x86_64-unknown-linux-gnu

installed targets for active toolchain
--------------------------------------

arm-unknown-linux-gnueabi
arm-unknown-linux-gnueabihf
x86_64-pc-windows-gnu
x86_64-unknown-linux-gnu

active toolchain
----------------

stable-x86_64-unknown-linux-gnu (default)
rustc 1.42.0 (b8cedc004 2020-03-09)
```

Thank you.

romainreignier, 2020-04-30:
Docker oneliner to reproduce the issue in a sane environment:

```
docker run --rm -it rust:1.43 sh -c "git clone https://github.com/adnanademovic/rosrust.git && cd rosrust && cargo run --example subscriber_node"
```

romainreignier, 2020-05-18:
Sorry, my last comment is bad because the Docker image used does not have ROS installed so for sure the message generation does not work :)

The correct would have been to install Rust in a ROS Docker image. And this one works:
```sh
docker run --rm -it ros:melodic bash -c "\
apt update && \
apt install -y curl && \
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- -y && \
source /root/.cargo/env && \
source /opt/ros/melodic/setup.bash && \
git clone https://github.com/adnanademovic/rosrust.git && \
cd rosrust && \
cargo run --example subscriber_node"
```

But I cannot manage to reproduce it on my computer.

This is the full error. Maybe I have too much message types installed and one of them generate an issue?

```
   Compiling rosrust v0.9.4 (/home/romain/projects/rust_ws/rosrust/rosrust)
error[E0277]: the trait bound `std_msgs::String: std::marker::Copy` is not satisfied
 --> /home/romain/projects/rust_ws/rosrust/target/debug/build/rosrust_msg-049699007daf2c72/out/messages.rs:3:9
  |
3 |         rosrust::rosmsg_include!(controller_manager_msgs/ReloadControllerLibraries,controller_manager_msgs/UnloadController,controller_manager_msgs/SwitchController,controller_manager_msgs/LoadController,controller_manager_msgs/ListControllerTypes,controller_manager_msgs/ListControllers,controller_manager_msgs/ControllersStatistics,controller_manager_msgs/HardwareInterfaceResources,controller_manager_msgs/ControllerStatistics,controller_manager_msgs/ControllerState,move_base_msgs/MoveBaseActionGoal,move_base_msgs/MoveBaseActionResult,move_base_msgs/MoveBaseResult,move_base_msgs/MoveBaseAction,move_base_msgs/MoveBaseFeedback,move_base_msgs/MoveBaseActionFeedback,move_base_msgs/MoveBaseGoal,uuid_msgs/UniqueID,gazebo_msgs/DeleteLight,gazebo_msgs/SpawnModel,gazebo_msgs/SetJointTrajectory,gazebo_msgs/ApplyJointEffort,gazebo_msgs/GetModelState,gazebo_msgs/GetModelProperties,gazebo_msgs/BodyRequest,gazebo_msgs/GetLinkProperties,gazebo_msgs/SetLinkProperties,gazebo_msgs/SetLinkState,gazebo_msgs/GetPhysicsProperties,gazebo_msgs/GetLinkState,gazebo_msgs/SetModelConfiguration,gazebo_msgs/GetWorldProperties,gazebo_msgs/GetJointProperties,gazebo_msgs/JointRequest,gazebo_msgs/DeleteModel,gazebo_msgs/SetJointProperties,gazebo_msgs/ApplyBodyWrench,gazebo_msgs/SetPhysicsProperties,gazebo_msgs/SetModelState,gazebo_msgs/GetLightProperties,gazebo_msgs/SetLightProperties,gazebo_msgs/ODEJointProperties,gazebo_msgs/WorldState,gazebo_msgs/ODEPhysics,gazebo_msgs/ContactsState,gazebo_msgs/LinkState,gazebo_msgs/ModelState,gazebo_msgs/LinkStates,gazebo_msgs/ModelStates,gazebo_msgs/ContactState,geometry_msgs/Wrench,geometry_msgs/AccelStamped,geometry_msgs/InertiaStamped,geometry_msgs/WrenchStamped,geometry_msgs/PolygonStamped,geometry_msgs/Vector3,geometry_msgs/Vector3Stamped,geometry_msgs/TwistStamped,geometry_msgs/QuaternionStamped,geometry_msgs/Pose2D,geometry_msgs/Transform,geometry_msgs/Inertia,geometry_msgs/AccelWithCovariance,geometry_msgs/Polygon,geometry_msgs/Point32,geometry_msgs/PointStamped,geometry_msgs/PoseStamped,geometry_msgs/Pose,geometry_msgs/Twist,geometry_msgs/AccelWithCovarianceStamped,geometry_msgs/Quaternion,geometry_msgs/Point,geometry_msgs/Accel,geometry_msgs/PoseWithCovariance,geometry_msgs/TwistWithCovarianceStamped,geometry_msgs/TransformStamped,geometry_msgs/PoseArray,geometry_msgs/TwistWithCovariance,geometry_msgs/PoseWithCovarianceStamped,roscpp_tutorials/TwoInts,rospy_tutorials/BadTwoInts,rospy_tutorials/AddTwoInts,rospy_tutorials/Floats,rospy_tutorials/HeaderString,geographic_msgs/GetGeographicMap,geographic_msgs/GetGeoPath,geographic_msgs/GetRoutePlan,geographic_msgs/UpdateGeographicMap,geographic_msgs/WayPoint,geographic_msgs/GeoPath,geographic_msgs/GeoPoseStamped,geographic_msgs/RouteSegment,geographic_msgs/MapFeature,geographic_msgs/GeographicMap,geographic_msgs/RouteNetwork,geographic_msgs/GeographicMapChanges,geographic_msgs/RoutePath,geographic_msgs/GeoPointStamped,geographic_msgs/GeoPoint,geographic_msgs/BoundingBox,geographic_msgs/GeoPose,geographic_msgs/KeyValue,fiducial_msgs/InitializeMap,fiducial_msgs/FiducialTransform,fiducial_msgs/FiducialMapEntryArray,fiducial_msgs/Fiducial,fiducial_msgs/FiducialMapEntry,fiducial_msgs/FiducialArray,fiducial_msgs/FiducialTransformArray,control_toolbox/SetPidGains,control_msgs/QueryCalibrationState,control_msgs/QueryTrajectoryState,control_msgs/GripperCommandResult,control_msgs/PointHeadAction,control_msgs/SingleJointPositionActionGoal,control_msgs/PointHeadGoal,control_msgs/JointTrajectoryResult,control_msgs/PointHeadActionResult,control_msgs/JointJog,control_msgs/PointHeadActionFeedback,control_msgs/JointTrajectoryActionResult,control_msgs/FollowJointTrajectoryActionFeedback,control_msgs/SingleJointPositionActionFeedback,control_msgs/JointTrajectoryActionGoal,control_msgs/JointTrajectoryAction,control_msgs/JointTolerance,control_msgs/GripperCommand,control_msgs/FollowJointTrajectoryActionGoal,control_msgs/GripperCommandGoal,control_msgs/GripperCommandAction,control_msgs/FollowJointTrajectoryActionResult,control_msgs/PointHeadResult,control_msgs/PointHeadFeedback,control_msgs/GripperCommandActionResult,control_msgs/FollowJointTrajectoryFeedback,control_msgs/SingleJointPositionAction,control_msgs/JointTrajectoryFeedback,control_msgs/JointControllerState,control_msgs/PointHeadActionGoal,control_msgs/SingleJointPositionResult,control_msgs/PidState,control_msgs/SingleJointPositionFeedback,control_msgs/FollowJointTrajectoryResult,control_msgs/Joi [... cut ...] ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `std::marker::Copy` is not implemented for `std_msgs::String`
  |
  = note: the `Copy` trait is required because the repeated element will be copied
  = note: this error originates in a macro (in Nightly builds, run with -Z macro-backtrace for more info)

error: aborting due to previous error

For more information about this error, try `rustc --explain E0277`.
error: could not compile `rosrust_msg`.

To learn more, run the command again with --verbose.
```

romainreignier, 2020-05-18:
I have found the culprit!
The package `variant_msgs` includes [std_msgs/String in a message definition](https://github.com/ANYbotics/variant/blob/393a0adacf4676c56782a12fff22285e80ba4f99/variant_msgs/msg/Test.msg#L8) instead of plain `string` type.
Actually it is a test message type that uses both string types by that is installed globally with the package.
This makes `rosrust_msgs` fail.

You can test we the following:
```sh
docker run --rm -it ros:melodic bash -c "\
apt update && \
apt install -y curl && \
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- -y && \
source /root/.cargo/env && \
source /opt/ros/melodic/setup.bash && \
git clone https://github.com/adnanademovic/rosrust.git && \
cd rosrust && \
cargo build && \
apt install ros-melodic-variant-msgs && \
cargo build"
```

So, even if it is a bad practice to use `std_msgs/String` in a message type, maybe `rosrust` could handle it?

romainreignier, 2020-05-18:
I have added this test in `msg.rs` but it passes:
```rust
    #[test]
    fn parse_variant_msgs_test() {
        let data = Msg::new(
            "variant_msgs/Test".try_into().unwrap(),
            include_str!("msg_examples/variant_msgs/msg/Test.msg"),
        )
        .unwrap();
        assert_eq!(data.path.package, "variant_msgs");
        assert_eq!(data.path.name, "Test");
        assert_eq!(
            data.fields,
            vec![
                FieldInfo {
                    datatype: DataType::RemoteStruct(MessagePath::new("std_msgs", "Header")),
                    name: "header".into(),
                    case: FieldCase::Unit,
                },
                FieldInfo {
                    datatype: DataType::U8(true),
                    name: "byte_constant".into(),
                    case: FieldCase::Const("42".to_string()),
                },
                FieldInfo {
                    datatype: DataType::I32,
                    name: "builtin_int".into(),
                    case: FieldCase::Unit,
                },
                FieldInfo {
                    datatype: DataType::Bool,
                    name: "builtin_boolean".into(),
                    case: FieldCase::Unit,
                },
                FieldInfo {
                    datatype: DataType::RemoteStruct(MessagePath::new("std_msgs", "Bool")),
                    name: "boolean".into(),
                    case: FieldCase::Unit,
                },
                FieldInfo {
                    datatype: DataType::String,
                    name: "builtin_string".into(),
                    case: FieldCase::Unit,
                },
                FieldInfo {
                    datatype: DataType::RemoteStruct(MessagePath::new("std_msgs", "String")),
                    name: "string".into(),
                    case: FieldCase::Unit,
                },
                FieldInfo {
                    datatype: DataType::I32,
                    name: "builtin_int_array".into(),
                    case: FieldCase::Array(3),
                },
                FieldInfo {
                    datatype: DataType::I32,
                    name: "builtin_int_vector".into(),
                    case: FieldCase::Vector,
                },
                FieldInfo {
                    datatype: DataType::RemoteStruct(MessagePath::new("std_msgs", "String")),
                    name: "string_array".into(),
                    case: FieldCase::Array(3),
                },
                FieldInfo {
                    datatype: DataType::RemoteStruct(MessagePath::new("std_msgs", "String")),
                    name: "string_vector".into(),
                    case: FieldCase::Vector,
                },
                FieldInfo {
                    datatype: DataType::Bool,
                    name: "builtin_boolean_array".into(),
                    case: FieldCase::Array(3),
                },
            ]
        );
        let dependencies = get_dependency_set(&data);
        assert_eq!(dependencies.len(), 3);
        assert!(dependencies.contains(&MessagePath::new("std_msgs", "Header")));
        assert!(dependencies.contains(&MessagePath::new("std_msgs", "Bool")));
        assert!(dependencies.contains(&MessagePath::new("std_msgs", "String")));
    }
```

adnanademovic, 2020-08-24:
I found the issue, and I'm resolving it. The fix should also resolve #140

adnanademovic, 2020-08-24:
This should be fixed in `rosrust 0.9.5` and `rosmsg 0.1.1`.

closed 2020-08-24 by adnanademovic

romainreignier, 2020-08-24:
Great! I have tried it and it is now working.
Thanks!