adnanademovic/rosrust #140: rosrust_msg fails to build

gstavrinos, 2020-07-31:
Hey,

First of all, thanks for the rust wrapper.
I had problems compiling the rosrust_msg library. I completely uninstalled rust and reinstalled but the problem persists.

The readable part of the error is the following:

```
error[E0277]: the trait bound `geometry_msgs::Pose: std::marker::Copy` is not satisfied
[...]
  = note: the `Copy` trait is required because the repeated element will be copied
  = note: this error originates in a macro (in Nightly builds, run with -Z macro-backtrace for more info)

error: aborting due to previous error

For more information about this error, try `rustc --explain E0277`.
error: could not compile `rosrust_msg`.
```

Any ideas or advice would be appreciated.

George

adnanademovic, 2020-08-24:
I cannot replicate your issue. Can you give more details on what you're doing?

adnanademovic, 2020-08-24:
Any further discussion will happen in the linked issue

closed 2020-08-24 by adnanademovic

gstavrinos, 2020-08-25:
Hey, thanks for taking a look into it. I eventually solved it uninstalling everything (including rust apart from rosrust) and then by compiling everything from source. This did not work the first time, but (for some weird reason) worked when I then used the cargo. Ugh!

I saw that you commited a fix though, so I guess I won't have to juggle around that problem again! Thanks!