adnanademovic/rosrust #145: How to use custom .srv

IceSentry, 2020-09-27:
I'm not sure I understand where I should put a custom .srv to be able to then use it with rosrust_msg. Is there a way to have it locally in a crate that uses rosrust?

pmirabel, 2020-11-09:
Hi,
I am currently trying to implement such a thing and integrate it with `catkin` and `roslaunch`.

You can find my work [on this example repo.](https://gitlab.com/pmirabel/rosrust-boilerplate/-/tree/custom_srv )  

By now, I'm stuck at runtime with this error after a `rosservice info (...)`  : 

```
contacting node http://localhost:34227/ ...
ERROR: remote call failed: Method not implemented
```

lucasw, 2020-11-09:
rosnode info also has the same `Method not implemented` error, but so far it doesn't look like it is hurting the ability of the node to run or communicate with other nodes on topics- so the service could be working fine, just no info?

pmirabel, 2020-11-09:
Thx for the answer. Unfortunately, I get an error wathever action I attempt with my rosservice.
Just to be 100% sure, if my service `request` is a `string` named `action`, the rosservice CLI call would be `rosservice call /my_service "action:'my string'"` ?

lucasw, 2020-11-09:
Does tab complete not work on the rust node `rosservice call`?  (It may be leveraging a rosservice info behind the scenes)

If you make an example C++ or python node in your project that implements the same service you can sort out what is not working in rosrust vs. a typo or some other mistake- and having example nodes in C++ or python alongside rust versions is super valuable for people new to rust (like me).

pmirabel, 2020-11-09:
> Does tab complete not work on the rust node `rosservice call`? (It may be leveraging a rosservice info behind the scenes)

Yep, autocompletion does not work either.

> If you make an example C++ or python node in your project (...)
Good one, thank you !

pmirabel, 2021-01-13:
Hi, I think that this issue could be closed :  
- create a service : https://gitlab.com/pmirabel/rosrust-boilerplate/-/tree/custom_srv  
- resolve ipv6 conflict : https://github.com/adnanademovic/rosrust/issues/46#issuecomment-726269177