adnanademovic/rosrust #149: Multiple subscriber for the same topic

OTL, 2020-10-11:
rosrust can't create multiple subscriber for one topic.
I want do like below, but it fails.
I tried to fix the problem, but rust is super difficult for me!!
I succeeded to this by chaning the caller_id and get each data stream, but it is not good solution.
Any one help me?

```
use crossbeam::channel::unbounded;
use rosrust;
use std::process::Command;

mod util;

mod msg {
    rosrust::rosmsg_include!(std_msgs / String);
}

#[test]
fn multiple_subscriber_to_rospy_publisher() {
    let _roscore = util::run_roscore_for(util::Language::Shell, util::Feature::Subscriber);
    let _publisher = util::ChildProcessTerminator::spawn(
        Command::new("rosrun").arg("rospy_tutorials").arg("talker"),
    );

    let (tx1, rx1) = unbounded();
    let (tx2, rx2) = unbounded();
    rosrust::init("hello_world_listener");
    let subscriber1 = rosrust::subscribe::<msg::std_msgs::String, _>("chatter", 100, move |data| {
        tx1.send(data.data).unwrap();
    })
    .unwrap();
    let subscriber2 = rosrust::subscribe::<msg::std_msgs::String, _>("chatter", 100, move |data| {
        tx2.send(data.data).unwrap();
    })
    .unwrap();
    util::test_subscriber(rx1, r"hello world (\d+\.\d+)", true, 20);
    util::test_subscriber(rx2, r"hello world (\d+\.\d+)", true, 20);

    assert_eq!(subscriber1.publisher_count(), 1);
    assert_eq!(subscriber2.publisher_count(), 1);
}
```

OTL, 2020-11-24:
I found @adnanademovic 's code [here](https://github.com/adnanademovic/rosrust/tree/multiple-subscribers)  and is older than my issue. Nice timing!

adnanademovic, 2020-11-24:
I haven't merged it yet since I haven't found the time to thoroughly test it. Test suite passes, but interop with ROS can sometimes be unpredictable. If you find issues with the code in that branch, let me know!

OTL, 2021-05-25:
@adnanademovic I confirmed that multiple subscriber works (just use 'subscribe' for the same topic).
I think we can close this issue, right?

OTL, 2021-07-13:
This is already fixed!
😊

closed 2021-07-13 by OTL