adnanademovic/rosrust #170: error: proc maro panicked

marloncori, 2022-04-03:
After adding the two dependencies according to your instructions and after using the "cargo build" command I got the following at the very end of the process: 

( ...)
"error: proc macro panicked
 --> C:\Users\NUC\Desktop\rust\publisher\target\debug\build\rosrust_msg-313b65941bcfc194\out/messages.rs:2:1
  |
2 | rosrust::rosmsg_include!(,IGNORE_BAD);
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  |
  = help: message: Failed to parse all message paths

error: could not compile `rosrust_msg` due to previous error
warning: build failed, waiting for other jobs to finish...
error: build failed"

How can we fix that?

By the way, I also cannot get how to run a publisher and subscriber or a client and a server at the same time, since there is no way to create a launch file to do so. Should I create two separate package with "cargo new <pkg name> --bin" and then add the two dependencies in each of the two generated Cargo.toml? And then if so, should I run "main.rs" in separate shells?

kalinkochnev, 2022-04-03:
I had the same issue earlier today and here's how I fixed it

**Specific the CMAKE_PREFIX_PATH or ROSRUST_MSG_PATH path variables.** 
1. To find where your ros installation is at, I would recommend searching for it with `find / -name "ros"`. For me, my ROS installation was located in /opt/ros/, however that may be different for you since I was running mine in a container. 

2. In order to get the prefix path, first run the setup script located here `. /opt/ros/<ros_distro_here>/setup.sh`. If you are using noetic, it sets a CMAKE_PREFIX_PATH variable. However, I had issues when running foxy instead since it assigned a different path variable instead of CMAKE_PREFIX_PATH. You should be able to compile and not get that error message anymore

3. When you run `rosrust`, this should then automatically discover where all the messages are that come with ROS. However, if you ran into the issue that there's a mismatch in type (when I tried with ROS foxy, it complained about there being different types for time) I decided to just specify only specific messages that I wanted with ROSRUST_MSG_PATH instead. 

4. To specify that I wanted only the `std_msgs`, I did `export ROSRUST_MSG_PATH=/opt/ros/<your_ros_distro>/share/std_msgs/`

5. Reminder, you can specify multiple paths if you separate each path with a colon `:`

Finally, in my main.rs I had 

```rust
mod msg {
    rosrust::rosmsg_include!(std_msgs/String);
}
```

I saw that there was also a pull request with a suggested warning to the user instead of panicking #165 to make it more clear what needs to be done. I highly suggest we merge this in since it was very confusing what was going on otherwise.

AthulMuralidhar, 2022-09-23:
+1 on this one -  I have the same issue as of now - looking forward to the release :)