adnanademovic/rosrust #181: Synchronising multiple topics

OmkarKabadagi5823, 2022-12-16:
I am working on a project where there is a need for the messages to be synced with time. I know `message_filters` exist for cpp and python, and I am guessing there is no such utility as of now available for rust. I wanted to know if there is any workaround for this.

Mitchelldscott, 2022-12-21:
save a history of timestamped messages from each topic and then compare the timestamps to find most recent/similar. It's definitely not pretty or efficient though.

OTL, 2022-12-22:
`tf_rosrust` has a similar implementation. (It may be not so efficient, too.)

https://github.com/smilerobotics/tf_rosrust/blob/main/src/tf_individual_transform_chain.rs

OmkarKabadagi5823, 2022-12-22:
Does this only work with TF messages? I am looking to synchronise `Image` msgs.

closed 2023-03-11 by OmkarKabadagi5823