adnanademovic/rosrust #183: subscribe_param

jobafr, 2023-01-27:
Hi,

I noticed there [seems to be some code already](https://github.com/adnanademovic/rosrust/blob/1728df1c46fb5572877b373ac18fd73e918cf782/rosrust/src/api/master.rs#L129) for subscribing parameters already, but it's not really usable in the public API. Subscribing to parameters is possible (although somewhat hacky) with rospy/roscpp already, but basically, roscore supports it.

Is there a specific reason subscribing to params is not included in rosrust's API? Would a PR be welcome?

adnanademovic, 2023-03-31:
The interface is there for completeness, but implementing it was a low priority. As you said, existing solutions are hacky, and it's all not extremely well documented.

PRs are welcome for sure.

jobafr, 2023-06-26:
Wow, this looks much better than the API I had since come up with! Feel free to close this.

closed 2023-08-07 by adnanademovic