adnanademovic/rosrust #188: rust analyzer not showing completion for custom messages but shows it for "built-in" messages like std_msgs/Header

noufelez, 2023-03-28:
Hi,

At our company, we're considering using rosrust for message generation and ser-deserialisation of messages in order to interop with a large codebase of c++ code.

Message generation works great, for example I could  generate an EnergySpeed message form a .msg file. But, when hovering over it or ctrl+clicking into it I don't get anything. But when I do it for Header (which is a ros builtin type) I get this 
[image]
and I can click on it which takes me to a lib.rs file located in the rosrust_msg crate which contains this code:

`include!(concat!(env!("OUT_DIR"), "/", "messages.rs"));`

Is this somehow a limitation of rust-analyzer or a way with which rosrust is currently designed ? @adnanademovic 

Thank you for the amazing work!

adnanademovic, 2023-04-01:
It might be a Rust Analyzer limitation. It might not be having a good time with finding or traversing to generated files within the `target` directory.

If it's any consolation, generating the documentation via `cargo doc --open` will give you documentation for all messages generated within `rosrust_msg`.

noufelez, 2023-04-03:
Thanks for your response! :)

closed 2023-04-03 by noufelez