adnanademovic/rosrust #189: First publish of publisher_node is lost

impactaky, 2023-04-01:
Hello,
Thanks to this repository I am able to communicate with ROS. 
However, it seems that the first send_msg is sometimes lost when the wait time after creating the publisher is not long enough.

## Reproduce procedure

After `rostopic echo`, run the example publisher_node.
Then output will like
```
$ rostopic echo chatter
data: "hello world from rosrust 1"
---
data: "hello world from rosrust 2"
---
data: "hello world from rosrust 3"
---
data: "hello world from rosrust 4"
---
data: "hello world from rosrust 5"
---
data: "hello world from rosrust 6"
...
```
```
$ cargo run --example publisher_node
[INFO @ examples/examples/publisher_node.rs:25]: Publishing: hello world from rosrust 0
[INFO @ examples/examples/publisher_node.rs:25]: Publishing: hello world from rosrust 1
[INFO @ examples/examples/publisher_node.rs:25]: Publishing: hello world from rosrust 2
[INFO @ examples/examples/publisher_node.rs:25]: Publishing: hello world from rosrust 3
[INFO @ examples/examples/publisher_node.rs:25]: Publishing: hello world from rosrust 4
[INFO @ examples/examples/publisher_node.rs:25]: Publishing: hello world from rosrust 5
[INFO @ examples/examples/publisher_node.rs:25]: Publishing: hello world from rosrust 6
...
```
The message `hello world from rosrust 0` seems to be missing.

## Environment

* Ubuntu 20.04
* ROS noetic
* rustc 1.70.0-nightly (a266f1199 2023-03-22)

adnanademovic, 2023-04-01:
I've recently become aware of the issue, thanks for the report.

The concern is that introducing a mandatory wait for subscribers is technically a breaking change, as misbehaving subscribers could cause a block.

I could add a `set_wait_for_subscribers` flag that will cause the initial message to wait for the initial subscribers to connect.

Then for a `0.10` I could make it the default.

impactaky, 2023-04-01:
Thank you.
I like `set_wait_for_subscribers`! 
For me, it seems enough that `set_wait_for_subscribers` is left default off and just written in the example. 

I was also advised by a colleague that sometimes it is enough to just set it to latched t to prevent lost first packets. (Perhaps both roscpp and rospy have this problem. I just happened to notice this issue  by using pubslisher with Lazy.)

adnanademovic, 2023-04-01:
There are use cases where it is important to provide every single messages. Playing back recorded data is a good example of that - think of `rosbag play`.

But for software robustness, your colleague is correct - setting latching is the correct way to go. That flag will make the publisher publish the last message it published to any subscribers that connect to it.

In ROS, the publisher/subscriber system should be used for a datapoint that continuously changes. In that sense, only the final value should be important. In fact, when the internal queues (whose sizes you set) get overflown, oldest messages in the queue are dumped - there is no waiting/blocking.

If you want a communication method that ensures everything is sent and received, use the client/service system. It has all the mechanisms for syncing up, and doesn't drop anything.

PR #190 (2023-04-01): Add wait for subscribers to publisher
Closes #189
closed 2023-04-01 by adnanademovic