adnanademovic/rosrust #208: Not compiling

guimath, 2024-09-09:
Hi i was trying to compile another lib using rosrust when it failed. I first tried to understand the issue so i cloned this repo and tried a local build directly. Still, it fails...

Here is of the errors: 
```
error[E0308]: mismatched types
   --> rosrust/src/api/ros.rs:71:79
    |
71  |                 .subscribe::<ClockMsg, _>("/clock", 1, move |v| clock.trigger(v.clock))
    |                                                                       ------- ^^^^^^^ expected `time::Time`, found `builtin_interfaces::Time`
    |                                                                       |
    |                                                                       arguments to this method are incorrect
    |
    = note: `builtin_interfaces::Time` and `time::Time` have similar names, but are actually distinct types
```
Also some problems in the rosrust/src/msg.rs, with the macro expanded: 
```
error[E0609]: no field `seq` on type `Header`
  --> rosrust/src/msg.rs:87:28
   |
87 |             if self.header.seq == 0 {
   |                            ^^^ unknown field
   |
   = note: available fields are: `stamp`, `frame_id`
```
I am using rustc 1.81.0 (on Linux Mint 21.1 Cinnamon) and don't really understand what's going on here. I am fully assuming that the problem is linked to my machine but don't really know where to look, any idea ?

Akumar201, 2024-09-20:
Hello , @guimath , As far as I understood from first error 
`error[E0308]: mismatched types
   --> rosrust/src/api/ros.rs:71:79
    |
71  |                 .subscribe::<ClockMsg, _>("/clock", 1, move |v| clock.trigger(v.clock))
    |                                                                       ------- ^^^^^^^ expected `time::Time`, found `builtin_interfaces::Time`
    |                                                                       |
    |                                                                       arguments to this method are incorrect
    |
    = note: `builtin_interfaces::Time` and `time::Time` have similar names, but are actually distinct types
`

`clock.trigger takes an argument of type  `time::Time` but got builtin_interfaces::Time``

The second error is saying your header does not have a function seq other , other fields are `stamp`, `frame_id`

can you please elaborate on how to replicate this error

guimath, 2024-09-21:
This is after no change to the code, simply running build on the freshly cloned repo.

For the second error it shows also without any changes but i just expand the macro to get a better understanding.
I understand the errors (although have dived too much in the code and so don't know what to change), what I don't get is how I am apparently the only one for who it is not working (I haven't tried it on another PC but I assume a published crate should be generally working)