adnanademovic/rosrust #210: Dependency xml-rs ^0.3.6 is yanked, breaking rosrust build

shioko-chan, 2024-12-23:
### **Problem**
The dependency `xml-rs = "^0.3.6"` required by `serde-xml-rs v0.2.1` has been yanked from `crates.io`, which breaks the build process when using `rosrust v0.9.11`. This makes it impossible to build projects that depend on `rosrust`.

### **Steps to Reproduce**
1. Create a new Rust project with the following `Cargo.toml`:
   ```toml
   [dependencies]
   rosrust = "0.9"
   rosrust_msg = "0.1.7"
2. Run cargo build or cargo run.

### **Error Message**
> error: failed to select a version for the requirement `xml-rs = "^0.3.6"`
> candidate versions found which didn't match: 0.8.24, 0.8.23, 0.8.22, ...
> location searched: crates.io index
> required by package `serde-xml-rs v0.2.1`
>     ... which satisfies dependency `serde-xml-rs = "^0.2.1"` of package `xml-rpc v0.0.12`
>     ... which satisfies dependency `xml-rpc = "^0.0.12"` of package `rosrust v0.9.11`
>     ... which satisfies dependency `rosrust = "^0.9.11"` of package `rosrust_msg v0.1.7`
>     ... which satisfies dependency `rosrust_msg = "^0.1.7"` of package `abysmal v0.1.0 (/root/abysmal)`
> if you are looking for the prerelease package it needs to be specified explicitly
>     xml-rs = { version = "0.8.15-cvss-cries-wolf" }

Would it be possible to release an updated version of rosrust with compatible dependencies?

MJohnson459, 2025-01-23:
The proper fix will need to filter through dependencies and updates but a workaround is to use a patch like this in your `Cargo.toml` file:

```toml
[patch.crates-io]
# Fixes https://github.com/adnanademovic/rosrust/issues/210
xml-rpc = { git = "https://github.com/locusrobotics/xml-rpc-rs", branch = "minimum-update" }
```

adnanademovic, 2025-02-03:
The dependency has been updated.
Thanks for the report.

closed 2025-02-03 by adnanademovic